OPCFoundation/UA-.NETStandard #183: MAXSubscriptionCount configuration property seems to be ignored

DarkoStanojevic, 2017-06-28:
Dear all,

I'm using the current verison of the OpcUa Standard-Library. For me it is not possible to increase the "MaxSubscriptionCount" (in the ServerConfiguration) higher than 10!

Nevertheless if am setting this property in the configuration lower (e.g. 5) or higher (e.g. 1000) the maximum of subscription I can add is 10. It seems that this property is ignored or is there any other thing to consider when increasing the amount of possible subscriptions on the server?

The request is sending the StatusCode "BadTooManySubscriptions" when the limit is exceeded.

Thanks a lot in advance and
best regards,
Darko

CodingHarry, 2017-07-03:
Same Problem here

mregen, 2017-07-12:
@DarkoStanojevic @CodingHarry I just verified MaxSubscriptionCount is working. Please use the Quickstarts.ReferenceServer.Config.xml as a reference. The setting might be ignored if its not at the right order in the file under ServerConfiguration. Please double check.

CodingHarry, 2017-07-13:
@mregen In my case this is not an issue based on the librarys server implementation. I am in use of an [IBH Link UA](https://www.ibhsoftec.com/IBH-Link-UA) as OPC-UA server. With [UA Expert](https://www.unified-automation.com/downloads/opc-ua-clients.html) i can do thousands of subscriptions without any special settings on the client (ua expert) side.

When i connect with a client implementation depending on this library i am not able to do more than 10 subscriptions.

I tried to set this on the client side...
`var configuration = new ApplicationConfiguration{
configuration.ServerConfiguration.MaxSubscriptionCount = 100;
}`
...without success.

Any suggestions?

DarkoStanojevic, 2017-07-18:
I have the same issue as "CodingHarry". I am using a real OpcUa-Server and the MaxSubscriptionCount-Setting seems to be completely ignored or overriden with the amount of 10.

Thanks in advance,
Darko

JayeshThamke, 2017-08-03:
Hi @mregen,
 I am also facing the same issue. I am developing an UA Client app and already have collected around 500 _VariableNodes_ to be subscribed. I am using C# code for configuring OPC UA (ClientAndServer) application. 
While subscribing more that 9 VariableNodes (on 10th Node subscription) it throws exception - Please refer the images attached.
Note - I have already set ApplicationConfiguration.ServerConfiguration.MaxSubscriptionCount = 10000.

[image]

Exception Information:
[image]

CallStackTrace:
[OPCUAFoundation_Subscription_StackTrace.txt](https://github.com/OPCFoundation/UA-.NETStandardLibrary/files/1197269/OPCUAFoundation_Subscription_StackTrace.txt)

JayeshThamke, 2017-08-06:
@mregen ,
My problem is solved now. 
I am currently using .NET Standard OPC UA library for developing OPC UA Client and The OPC UA Server is developed using Unified Automation GmbH .NET sdk. BadTooManySubscription exception message is displayed in a ServiceResult Exception when subscription for more that 10 VariableNodes using foreach loop.  
I have observed in my Server configuration, MaxSubscriptionCount was not correctly configured. I have now configured it (MaxSubscriptionCount = 10000 in OPC UA Server) and OPC UA Client (.NET Standard OPC UA library) does not throw any exception. Please check the OPC UA Server's Configuration setting.

@CodingHarry , 
I am successful in subscribing nodes (VariableNodes more that 100 ) in UaExpert ("**Manually**") . Please try the recommended changes in Application Configuration Setting above. For me Node subscription as well as data change notifications work fine.

AlinMoldovean, 2017-08-07:
Hello @JayeshThamke,

Your problem is caused because you are not creating the monitored items in a correct way.
You add a single monitored item per subscription in the server, i.e. you are creating a subscription for each monitored item.
Setting the MaxSubscriptionCount configuration parameter to a high value is not the best fix. It will generate too much trafic.

You need to create a single subscription per sampling group (e.g. 1000 ms) and add all the related monitored items inside that subscription.

In your SubscribeNode() method, the subscription object should be created only once and reused afterwards.

JayeshThamke, 2017-08-07:
Hello @AlinMoldovean ,

Thanks for reply and correction, I appreciate your efforts.

If I understood you correctly then I should create one Subscription (may be one per my device connected to server) and all VariableNodes should be added a new MonitoredItems in the created subscription. 

Could you please explain more about the "sampling group" you have mentioned above ?

@AlinMoldovean - In your SubscribeNode() method, the subscription object should be created only once and reused afterwards.

Me - Does the **reusing already created subscription** object mean **reusing** the subscription for adding more MonitoredItems?

Thanks,
Jayesh

AlinMoldovean, 2017-08-08:
Hello @JayeshThamke,

Yes, you can create a single Subscription object and add all your monitored items inside.
Usually one subscription per client session is enough.

JayeshThamke, 2017-08-08:
Hello @AlinMoldovean  
Thanks for support!

closed 2017-08-10 by mregen