OPCFoundation/UA-.NETStandard #186: CertificateFactory.CreateCertificate only first alternate name is used

ThomasLevering, 2017-07-03:
CertificateFactory.CreateCertificate only first alternate name is used

I need the hostname and all local ip adresses in the certifikate

MyCode
List<string> serverDomainNames = new List<string>();
string HostName = System.Net.Dns.GetHostName();
serverDomainNames.Add(HostName);
System.Net.IPHostEntry hostInfo = System.Net.Dns.GetHostEntry(HostName);
foreach (System.Net.IPAddress ip in hostInfo.AddressList)
  {
     if (ip.AddressFamily == System.Net.Sockets.AddressFamily.InterNetwork)
  {
    serverDomainNames.Add(ip.ToString());                    }
  }
 CertificateFactory.CreateCertificate(... serverDomainNames...    )

StackCode
Stack/Opc.Ua.Core/Security/Certificates/CertificateFactory.cs

                // subject alternate name
                cg.AddExtension(X509Extensions.SubjectAlternativeName, false,
                    new GeneralNames(new GeneralName[] {
                    new GeneralName(GeneralName.UniformResourceIdentifier, applicationUri),
                    new GeneralName(GeneralName.DnsName, domainNames[0])}));

mregen, 2017-07-03:
@ThomasLevering Good point, could you provide the fix as pull request?

mregen, 2017-09-27:
Fixed in #254

ThomasLevering, 2017-09-29:
with Version 254 nothing has changed

ip addresses can also be added here:

                // subject alternate name
                int subjectAlternativeCount = domainNames.Count + 1;
                if (ipAdresses != null)
                    subjectAlternativeCount = subjectAlternativeCount + ipAdresses.Count;
                GeneralName[] subjectAlternativeNames = new GeneralName[subjectAlternativeCount];
                subjectAlternativeNames[0] = new GeneralName(GeneralName.UniformResourceIdentifier, applicationUri);
                for (int n = 0; n < domainNames.Count; n++)
                {
                    subjectAlternativeNames[n + 1] = new GeneralName(GeneralName.DnsName, domainNames[n]);
                }
                if (ipAdresses != null)
                {
                    for (int n = 0; n < ipAdresses.Count; n++)
                    {
                        subjectAlternativeNames[n + domainNames.Count + 1] = new GeneralName(GeneralName.IPAddress, ipAdresses[n]);
                    }
                }
                cg.AddExtension(X509Extensions.SubjectAlternativeName, false, new GeneralNames(subjectAlternativeNames));

 //cg.AddExtension(X509Extensions.SubjectAlternativeName, false,
 //    new GeneralNames(new GeneralName[] {
 //    new GeneralName(GeneralName.UniformResourceIdentifier, applicationUri),
 //    new GeneralName(GeneralName.DnsName, domainNames[0])}));

[CertificateFactory.zip](https://github.com/OPCFoundation/UA-.NETStandardLibrary/files/1343714/CertificateFactory.zip)

mregen, 2017-09-29:
Stay tuned, #254 is still in code review. please check the fix here: https://github.com/mregen/UA-.NetStandardLibrary/commit/6cdc0d488c45213de0b771c5c2a416130cb4e39d#diff-c655c560b411619295b996543320f728

mregen, 2017-10-06:
fix is merged to master

closed 2017-10-09 by ThomasLevering