OPCFoundation/UA-.NETStandard #187: Compatibility with .Net Core 2.0

gava88, 2017-07-03:
We are making some tests on our embedded device (armhf - runtime ubuntu.14.04-arm, but also on my Windows 7 machine) and I'm facing some issues with the re-implemented Socket’s Task-based SendAsync and ReceiveAsync operations in .Net Core 2.0. [PR dotnet/corefx #16502](https://github.com/dotnet/corefx/pull/16502) 

.Net Core version:

> .NET Command Line Tools (2.0.0-preview2-006497)
> 
> Product Information:
>  Version:            2.0.0-preview2-006497
>  Commit SHA-1 hash:  06a2093335
> 
> Runtime Environment:
>  OS Name:     Windows
>  OS Version:  6.1.7601
>  OS Platform: Windows
>  RID:         win7-x64
>  Base Path:   C:\Program Files\dotnet\sdk\2.0.0-preview2-006497\
> 
> Microsoft .NET Core Shared Framework Host
> 
>   Version  : 2.0.0-preview2-25407-01
>   Build    : 40c565230930ead58a50719c0ec799df77bddee9

Opc.Ua.Core.csproj:

```
<Project Sdk="Microsoft.NET.Sdk">

  <PropertyGroup>
    <TargetFramework>netcoreapp2.0</TargetFramework>
    <!--<RuntimeFrameworkVersion>2.0.0-beta-001737-00</RuntimeFrameworkVersion>-->
    <RuntimeIdentifiers>win8-arm;ubuntu.14.04-arm;ubuntu.16.04-arm</RuntimeIdentifiers>
    <DefineConstants>$(DefineConstants);NET_STANDARD</DefineConstants>
    <AssemblyName>Opc.Ua.Core</AssemblyName>
    <PackageId>Opc.Ua.Core</PackageId>
    <DefaultLanguage>en-US</DefaultLanguage>
    <!--<NetStandardImplicitPackageVersion>1.6.0</NetStandardImplicitPackageVersion>-->
    <GenerateAssemblyTitleAttribute>false</GenerateAssemblyTitleAttribute>
    <GenerateAssemblyDescriptionAttribute>false</GenerateAssemblyDescriptionAttribute>
    <GenerateAssemblyCompanyAttribute>false</GenerateAssemblyCompanyAttribute>
    <GenerateAssemblyProductAttribute>false</GenerateAssemblyProductAttribute>
    <GenerateAssemblyCopyrightAttribute>false</GenerateAssemblyCopyrightAttribute>
    <GenerateAssemblyVersionAttribute>false</GenerateAssemblyVersionAttribute>
    <GenerateAssemblyFileVersionAttribute>false</GenerateAssemblyFileVersionAttribute>
  </PropertyGroup>

  <ItemGroup>
    <EmbeddedResource Include="Stack\Generated\Opc.Ua.PredefinedNodes.uanodes;Types\Utils\LocalizedData.txt" />
  </ItemGroup>

  <ItemGroup>
    <PackageReference Include="Portable.BouncyCastle" Version="1.8.1.2" />
    <PackageReference Include="Newtonsoft.Json" Version="10.0.3" />
    <PackageReference Include="System.Data.Common" Version="4.3.0" />
    <PackageReference Include="System.Private.ServiceModel" Version="4.3.0" />
    <PackageReference Include="System.ServiceModel.Primitives" Version="4.3.0" />
  </ItemGroup>

  <ItemGroup>
    <PackageReference Include="Microsoft.AspNetCore.Server.Kestrel" Version="1.0.3" />
    <PackageReference Include="Microsoft.AspNetCore.Server.Kestrel.Https" Version="1.0.3" />
  </ItemGroup>

</Project>

```

NetCoreConsoleClient.csproj:

```
<Project Sdk="Microsoft.NET.Sdk">

  <PropertyGroup>
    <TargetFramework>netcoreapp2.0</TargetFramework>
    <AssemblyName>NetCoreConsoleClient</AssemblyName>
    <OutputType>Exe</OutputType>
    <PackageId>NetCoreConsoleClient</PackageId>
    <GenerateAssemblyConfigurationAttribute>false</GenerateAssemblyConfigurationAttribute>
    <GenerateAssemblyCompanyAttribute>false</GenerateAssemblyCompanyAttribute>
    <GenerateAssemblyProductAttribute>false</GenerateAssemblyProductAttribute>
  </PropertyGroup>

  <ItemGroup>
    <ProjectReference Include="..\..\SDK\Opc.Ua.Client\Opc.Ua.Client.csproj" />
    <ProjectReference Include="..\..\..\Stack\Opc.Ua.Core\Opc.Ua.Core.csproj" />
  </ItemGroup>

</Project>
```

In particular, the console output generated by the NetCoreConsoleClient is:

> 11:52:50.090 GetEndpoints Called. RequestHandle=1, PendingRequestCount=1
> 11:52:50.142 Channel 0 in Connecting state.
> 11:52:50.264 Channel 0: SendHelloMessage()
> 11:52:50.273 Bytes read: 8
> 11:53:05.143 *** EXCEPTION*** BadRequestTimeout
> 11:53:05.143 *** EXCEPTION*** BadRequestTimeout
> Generated exception: 'Opc.Ua.ServiceResultException' in Opc.Ua.Core.dll
> Generated exception: 'Opc.Ua.ServiceResultException' in Opc.Ua.Core.dll
> 11:53:05.256 GetEndpoints Completed. RequestHandle=1, PendingRequestCount=0, StatusCode=Bad
> Eccezione generata: 'Opc.Ua.ServiceResultException' in NetCoreConsoleClient.dll
> 11:53:05.276 Channel 0: Close
> 11:53:05.280 Channel 0 in Closed state.
> 11:53:05.282 CLIENTCHANNEL SOCKET CLOSED: 038A2A76, ChannelId=0
> Generated exception: 'System.AggregateException' in System.Private.CoreLib.dll

Surely I could have made mistakes, but has anyone else tried or had the same problems with .Net Core 2.0?
The same example works perfectly with .Net Core1.1
I'd like to know what your plans are about integrating the stack with .Net Core2.0/.Net Standard2.0?

Thanks in advance.

barnstee, 2017-07-03:
As soon as .NetCore 2.0 comes out of preview, we'll consider upgrading to it.

WillooWisp, 2017-07-13:
I think you should update it ASAP, cannot be that hard for you. More and more NuGet packages from Microsoft only supports .NET Standard 2.0 so .NET Core 2.0 is sometimes necessary, especially if targeting ARM. So please give this some effort, or please tell us what to modify our self to get it up and running on .NET Core 2 Preview 2.

WillooWisp, 2017-07-13:
Here is part of the callstack for the exception:
'Opc.Ua.ServiceResultException: Error establishing a connection: BadRequestTimeout
   at Opc.Ua.Bindings.ChannelAsyncOperation`1.End(Int32 timeout, Boolean throwOnError) in OpcFoundation.OpcUA\Opc.Ua.Core\Stack\Tcp\ChannelAsyncOperation.cs:line 168
   at Opc.Ua.Bindings.UaSCUaBinaryClientChannel.EndSendRequest(IAsyncResult result) in OpcFoundation.OpcUA\Opc.Ua.Core\Stack\Tcp\UaSCBinaryClientChannel.cs:line 303
   at Opc.Ua.DiscoveryClient.GetEndpoints(RequestHeader requestHeader, String endpointUrl, StringCollection localeIds, S
tringCollection profileUris, EndpointDescriptionCollection& endpoints) in OpcFoundation.OpcUA\Opc.Ua.Core\Stack\Generated\Opc.Ua.Client.cs:line 4821
   at Opc.Ua.DiscoveryClient.GetEndpoints(StringCollection profileUris) in OpcFoundation.OpcUA\Opc.Ua.Core\Stack\Client\DiscoveryClient.cs:line 75

cpipero, 2017-07-13:
I second @WillooWisp comment. 2.0 is already extensively being used by a lot of dependencies (I too am doing .NET Core on ARM). I'm sure the community can help the porting with a few direction. Thanks again.

barnstee, 2017-07-14:
It's not a question of "how hard can it be?" It's a question of being able to support all the dependencies we have and all of the runtimes, plus it's question of prioritizing this work vs all the other feature requests, plus being able to test it all. .NetCore is one of 3 runtimes we support. Moving from Dotnetcore 1.0 to 1.1 was also non trivial! We've made the decision to wait tuntil dotnet 2.0 is officially released.

viados, 2017-08-23:
This issue is still present in the latest (final) release of .NET Core 2.0 (see #234) and is only present in case of .NET Core 2.0 is used. This issue is also reproducable on Win10 (x64, compiled with dotnet in VS2017) and OSX10.12 (x64, compiled with dotnet in VS for Mac).

WillooWisp, 2017-08-24:
Not prioritizing .NET Core 2.0 seems pretty weird, since everyone will want to upgrade to that. Currently this library is the only thing blocking us from going forward. Please share some information with us, for when we can expect a version, beta or whatever. Please also share some information for how we can patch it our self, the timeout issue is a blocker.

dwcy, 2017-09-11:
Any updates on this? or did someone manage to fix it?

 And I agree that focus on dotnetcore 2.0 would be top priority.

mregen, 2017-09-11:
@dwcy --> #237 in code review

mregen, 2017-09-20:
Merged to master

closed 2017-09-20 by mregen

venkatrv, 2018-11-02:
Hi all,
I am also looking for .NET core based OPC library. Is it available now? please let me know