OPCFoundation/UA-.NETStandard #189: Re-enable HTTPS transport for .Net

barnstee, 2017-07-04:
Once this is fixed: https://github.com/aspnet/Hosting/issues/1115, reenable HTTPS listener/transport in .Net samples.

barnstee, 2017-08-07:
Looks like this will be fixed for packages.config-based projects in VS2017 update1: http://blog.nuget.org/20170316/NuGet-now-fully-integrated-into-MSBuild.html

mregen, 2018-02-06:
HTTPS server state as of today:
1. Kestrel 1.1/netcoreapp1.1 servers in .Net Core samples: ok
2. Kestrel 1.1 servers in .Net 4.6 samples: broken unless user copies matching libuv.dll in release dir
3. Kestrel 2.0 in netcoreapp 2.0 server samples: only responds to some clients, not to UA samples at all (testing fix)

HTTPS client state:

4. bug: client gets stuck if subscription is active by nesting too many calls (testing fix, use async callback)
5. https client needs server cert in a trusted store or it refuses connection (testing fix to auto accept server certs)
6. https connection from .Net Core apps works if server accepts incoming connection and server cert is trusted
7. runtime issues with .Net 4.6. apps (cannot find diagnostics dll) --> workaround is to manually add nuget package of System.Net.http 4.3.3

mregen, 2018-02-06:
Fix in #342 for all but the old libuv.dll problem...

mregen, 2018-02-14:
https client still broken in 1.3.351.7, TimeOut method throws on second call to PostAsync...

mregen, 2018-02-15:
workaround in https://github.com/aspnet/KestrelHttpServer/issues/1652 & https://github.com/aspnet/KestrelHttpServer/issues/1292

closed 2018-02-21 by mregen