OPCFoundation/UA-.NETStandard #200: SuppressHostNameInvalid flag not implemented in CheckCertificateDomain()

barnstee, 2017-07-13:
This leads to incompatibilities for example with SIEMENS S7 OPC UA servers, who have an invalid hostname in the UA certificate. A client should at least have the option to ignore this error.

mregen, 2017-07-14:
Currently there is an option 'checkDomain' with session.open, it allows to ignore the hostname check.

Currently not supported is to specify e.g.:

```
    <TrustedPeerCertificates>
      <StoreType>Directory</StoreType>
      <StorePath>%CommonApplicationData%\OPC Foundation\CertificateStores\UA Applications</StorePath>
      <!-- Mask 0x2: SuppressHostNameInvalid -->
      <ValidationOptions>2</ValidationOptions>
    </TrustedPeerCertificates>
```

This ValidationOptions is currently ignored by clients and servers...
The tricky part is to associate the cert with the right store, which all may have different ValidationOptions, or simpler just use the ValidationOption in TrustedPeerCertificates for all validation checks. Tbd.

mregen, 2018-01-25:
#333 should address this problem. S7 has a bunch of ip addresses in the cert which are properly handled with this fix

closed 2018-01-25 by mregen