OPCFoundation/UA-.NETStandard #201: Ua Server closed or disconnected (event) notification in Ua Client does'nt call SessionClosing EventHandler mehtod

JayeshThamke, 2017-07-14:
-A session with Ua Server is already established with Ua Client.
-When the Ua Server is closed it does not call event handler method 

`OnSessionClosingHandler(object sender, EventArgs e)`

My implementation is as follows:

`Session session = await Session.Create(_AppConfiguration,
configuredEndpoint,
true,
true,
_AppConfiguration.ApplicationName,
15000,
new UserIdentity(new AnonymousIdentityToken()),
null);
session.SessionClosing += OnSessionClosingHandler;
` 
`private static void OnSessionClosingHandler(object sender, EventArgs e)
{ //have inserted breakpoint to debug the execution
    //some handling 
}`

Please also guide me if this is a correct approach to handle when Ua Server is closed/closing.
Thanks,
Jayesh

K47S, 2017-07-18:
You have to listen to the Session.KeepAlive event. Session closing is only called when the server intended to close the session.

JayeshThamke, 2017-07-18:
Thank for reply @K47S 
Your second statement is not clear to me.
Does it mean, I should implement the session closing handling in Session.KeepAlive event handler method?

Do you from

> when the server intended to close the session.

statement mean that only closing Ua Server application does not close the Session with client and it has to be explicitly closed from Ua Server side?

Could you please elaborate it?

Thanks a lot,
Jayesh

K47S, 2017-07-18:
No, on KeepAlive you should try to reconnect/recreate the session...

It depends on the implementation of the server if it closes the session and therefor raise the Session.OnSessionClosingHandler. If you forcefully close the server its very likely that the event will not fire...

closed 2017-08-06 by JayeshThamke

jwanna, 2022-04-11:
@JayeshThamke 
 I have the same issue of you ! 
 Did you find a solution to call the SessionClosing when the Ua Server is disconnected or closed ? 

I didn't get the answer of @MD-V 

Regards,