OPCFoundation/UA-.NETStandard #203: StoreType Directory for ApplicationCertificate thows on second start of server

hansgschossmann, 2017-07-17:
Here the snippet of the ApplicationCertificate configuration:
    <ApplicationCertificate>
      <StoreType>Directory</StoreType>
      <StorePath>d:/temp/Shared/CertificateStores/own</StorePath>
      
      <SubjectName>UndefinedStationName</SubjectName>
    </ApplicationCertificate>

If i start the server, it works fine on the first start and it creates the certificates in the directories. On the second start of the server, it throws an exception with the following call stack:

>	Opc.Ua.Core.dll!Opc.Ua.Bindings.UaSCUaBinaryChannel.RsaPkcs15_Sign(System.ArraySegment<byte> dataToSign, System.Security.Cryptography.X509Certificates.X509Certificate2 signingCertificate, System.Security.Cryptography.HashAlgorithmName algorithm) Line 87	C#
 	Opc.Ua.Core.dll!Opc.Ua.Bindings.UaSCUaBinaryChannel.Sign(System.ArraySegment<byte> dataToSign, System.Security.Cryptography.X509Certificates.X509Certificate2 senderCertificate) Line 966	C#
 	Opc.Ua.Core.dll!Opc.Ua.Bindings.UaSCUaBinaryChannel.WriteAsymmetricMessage(uint messageType, uint requestId, System.Security.Cryptography.X509Certificates.X509Certificate2 senderCertificate, System.Security.Cryptography.X509Certificates.X509Certificate2 receiverCertificate, System.ArraySegment<byte> messageBody) Line 578	C#
 	Opc.Ua.Core.dll!Opc.Ua.Bindings.UaSCUaBinaryClientChannel.SendOpenSecureChannelRequest(bool renew) Line 481	C#
 	Opc.Ua.Core.dll!Opc.Ua.Bindings.UaSCUaBinaryClientChannel.ProcessAcknowledgeMessage(System.ArraySegment<byte> messageChunk) Line 446	C#
 	Opc.Ua.Core.dll!Opc.Ua.Bindings.UaSCUaBinaryClientChannel.HandleIncomingMessage(uint messageType, System.ArraySegment<byte> messageChunk) Line 693	C#
 	Opc.Ua.Core.dll!Opc.Ua.Bindings.UaSCUaBinaryChannel.OnMessageReceived(Opc.Ua.Bindings.IMessageSocket source, System.ArraySegment<byte> message) Line 289	C#
 	Opc.Ua.Core.dll!Opc.Ua.Bindings.TcpMessageSocket.DoReadComplete(System.Net.Sockets.SocketAsyncEventArgs e) Line 607	C#
 	Opc.Ua.Core.dll!Opc.Ua.Bindings.TcpMessageSocket.OnReadComplete(object sender, System.Net.Sockets.SocketAsyncEventArgs e) Line 502	C#
 	System.Net.Sockets.dll!System.Net.Sockets.SocketAsyncEventArgs.OnCompleted(System.Net.Sockets.SocketAsyncEventArgs e)	Unknown
 	System.Net.Sockets.dll!System.Net.Sockets.SocketAsyncEventArgs.ExecutionCallback(object state)	Unknown
 	System.Private.CoreLib.ni.dll!System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, object state)	Unknown
 	System.Net.Sockets.dll!System.Net.Sockets.SocketAsyncEventArgs.FinishOperationSuccess(System.Net.Sockets.SocketError socketError, int bytesTransferred, System.Net.Sockets.SocketFlags flags)	Unknown
 	System.Net.Sockets.dll!System.Net.Sockets.SocketAsyncEventArgs.CompletionPortCallback(uint errorCode, uint numBytes, System.Threading.NativeOverlapped* nativeOverlapped)	Unknown
 	System.Threading.Overlapped.dll!System.Threading.ThreadPoolBoundHandleOverlapped.CompletionCallback(uint errorCode, uint numBytes, System.Threading.NativeOverlapped* nativeOverlapped)	Unknown
 	System.Private.CoreLib.ni.dll!System.Threading._IOCompletionCallback.PerformIOCompletionCallback(uint errorCode, uint numBytes, System.Threading.NativeOverlapped* pOVERLAP)	Unknown

Here are the exception details:

System.Security.Cryptography.CryptographicException occurred
  HResult=0x80131500
  Message=OpenCSP failed with error code 2148073494.
  Source=<Cannot evaluate the exception source>
  StackTrace:
   at Internal.NativeCrypto.CapiHelper.CreateProvHandle(CspParameters parameters, Boolean randomKeyContainer)

The source code which is shown is:

        /// <summary>
        /// Creates an RSA PKCS#1 v1.5 signature of a hash algorithm for the stream.
        /// </summary>
        private static byte[] RsaPkcs15_Sign(
            ArraySegment<byte> dataToSign,
            X509Certificate2 signingCertificate,
            HashAlgorithmName algorithm)
        {
            // extract the private key.
           using (RSA rsa = signingCertificate.GetRSAPrivateKey())
            {
                if (rsa == null)
                {
                    throw ServiceResultException.Create(StatusCodes.BadSecurityChecksFailed, "No private key for certificate.");
                }

                // create the signature.
                return rsa.SignData(dataToSign.Array, dataToSign.Offset, dataToSign.Count, algorithm, RSASignaturePadding.Pkcs1);
            }
        }

The GetRSAPrivateKey() is causing the exception.

barnstee, 2017-07-17:
This is not repro for me on master with the NetCoreConsoleServer (I assume this is with .NetCore?). From the trace, it looks like this is happening when connecting to the server?

mregen, 2017-09-15:
@hansgschossmann As discussed, this issue is due to the unusual initialization of cert CN in the factory simulation code, won't fix. Workaround: use X509 store.

closed 2017-09-15 by mregen