OPCFoundation/UA-.NETStandard #208: How to auto-browse in Ua Client the dynamically creating and deleting node/s from Ua Server

JayeshThamke, 2017-07-17:
Hello,
In my Ua Server implementation there are some use cases where user can dynamically add and remove the Nodes; those cases are implemented and works fine. How to implement Auto-browse feature in Ua Client? Is it already implemented in Sample example? Please share the link. 
I have some hint such as referencing _Node Version_ property and issuing _ModelChangeEvent_ in Ua Server but I have no no idea how it is handled in Ua Client.

Thank you,

K47S, 2017-07-18:
You could try to copy some code from the AggregationServer which browses the other server and "copies" the whole addresspace:
[https://github.com/OPCFoundation/UA-.NETStandardLibrary/tree/master/SampleApplications/Workshop/Aggregation](https://github.com/OPCFoundation/UA-.NETStandardLibrary/tree/master/SampleApplications/Workshop/Aggregation)

JayeshThamke, 2017-08-06:
Hi @K47S,
Can you point out what portion of code from Aggregation Server does this functionality? Thanks,

AlinMoldovean, 2017-09-27:
Hello @JayeshThamke,

This can be implemented by reporting a ModelChangeEvent event in the Server when the address space changes.
In order to get notified, the client will have to subscribe to Event notifications for this type of events.
It will later refresh the nodes using Browse.

For the moment this feature is not implemented in the .NET Standard Server library, so the server is not automatically triggering ModelChange events.
As far as I know there is no example for this in a sample application.

closed 2017-09-27 by AlinMoldovean

JayeshThamke, 2017-09-27:
To see the change in Server's Information Model, I have subscribed NodeVersion Variable Node in client, and handled in special MonitoredItem Notification handler for IM change. At least for my academic project it is working fine.

Please see #242