OPCFoundation/UA-.NETStandard #209: How client restore session after server restart

MRK89TP, 2017-07-18:
Hi, how can the client restore the session (with subscription and monitoredItem inside) on Server after Server restart?
Without create new session, subscription and monitoredItem.
I tried  session.Reconnect()  but it throw BadSessionIdInvalid exception.
And i tried session.Recreate(session) , in this case ServerState becomes Running, but client not receices event on monitoreItem “Notification” event.
Thanks.

K47S, 2017-07-18:
You have to recreate the subscription:

` if (subscription.Created)
                {
                    subscription.Delete();
                }
              
                session.AddSubscription(subscription);
                subscription.Create();`

MRK89TP, 2017-07-18:
Thanks for answare!

On the one hand subscritpion.Delete(true) resolve the problem of switching monitoredItem.Status.Created to false that i've described here  https://opcfoundation.org/forum/opc-ua-standard/restore-connection-after-server-restart/#p699

On the other hand I received BadSessionIdInvalid on subscription.Create().

Obviously the server loses the old session after restart, Can this be a problem?

mregen, 2017-09-27:
@MRK89TP if you receive BadSessionIdInvalid on subscription.Create() the session must be recreated. Reconnect session works only for existing sessions but not if the server restarted.

AlinMoldovean, 2017-09-27:
Hello @MRK89TP,

An implementation example can be found in the SessionReconnectHandler class of the Client library.

closed 2017-09-27 by AlinMoldovean