OPCFoundation/UA-.NETStandard #212: Utils.Tracing(.Instance) ~ UA-.NET

kvloover, 2017-07-27:
I'm currently investigating the use of the UA-.NETStandardLibrary over the UA-.NET library.
However I see that the Utils.Tracing.Instance event raising is not implemented in the StandardLibrary.

Logging is something very important and if possibly I would like to log in the manner I'm logging other things. The UA-.NET library allows me to do that by disabling the tracing and handling the event to determine the necessary tracing myself.

Is there a reason this has been removed in the standardlibrary implementation ?

mregen, 2017-08-01:
@kvloover I presume the tracing nuget package wasn't available until recently. Please try this branch:
https://github.com/mregen/UA-.NetStandardLibrary/tree/trace_instance
and let me know if it works. We can bring it back.

AlinMoldovean, 2017-09-28:
Hello @kvloover ,

We will try to implement a custom tracing event in the same was as in UA-.NET repository.
It will allow to use a custom logging system in applications.

Thanks,
Alin

commit 366ff5ca (2017-09-29): Fixes #212 - Tracing event ported from UA .NET repository
closed 2017-09-29 by mregen