OPCFoundation/UA-.NETStandard #216: Keypair was disposed errors using latest mono

Savron, 2017-08-04:
Using the latest stable version of mono the UA-.NETStandardLibrary can now be used with only a few minor tweaks.

One thing that is still causing issues and took me a while to debug was a persistent "Keypair was disposed" error.

I finally tracked the problem to the way OPC UA calls the GetRSAPublicKey() function.

For example in CertificateValidator.cs and some other places it is called this way "using (RSA rsa = certificate.GetRSAPublicKey())".

This doesn't cause any issues with the Microsoft .NET framework but in mono GetRSAPublicKey() uses the certificate.PublicKey.Key property, which generates a RSA object using "new" the first time it is accessed and internally saves a reference to it, which it will return on any subsequent access to the .Key property, instead of decoding the key anew.
This is also the object returned by GetRSAPublicKey().

Calling GetRSAPublicKey() with the "using" keyword causes this object to be disposed and any subsequent call to GetRSAPublicKey() or any other function which accesses the certificate.PublicKey.Key property will then get a reference to the now disposed object it had stored internally thus causing the "Keypair was disposed" errors.

I don't know the rationale it was called with the "using" keyword in this context, but removing it from all calls to GetRSAPublicKey() will allow everything to work without any problems.

barnstee, 2017-08-04:
That's a great find!

dacolgit, 2017-08-08:
@Savron Is security turned On in your test?
I'm trying to implement an OPC UA client on Android using Xamarin. If security is turned Off the client works and can connect to a server. When I turn On security It doesn't work anymore.
Xamarin uses the Mono library.

The problem is in CertificateFactory.cs in the CreateCertificateFromPKCS12 method.

The following code create a certificate with a private key

`certificate = new X509Certificate2(
                    rawData,
                    (password == null) ? String.Empty : password,
                    storageFlags[flagsRetryCounter]);`

but 

`using (RSA rsa = certificate.GetRSAPrivateKey()) { }`

cannot access to the private key, it doesn't return any exception, it just set the 'HasPrivateKey' field in the certificate to false.

stefanocadoaleph, 2017-12-09:
Hi, I has the same issues while trying to run opc ua server on mono on yocto linux build (Siemes IOT2000). Here it is my solution: https://github.com/aleph-solutions/UA-.NETStandard/commit/fd730f2cce4fc1d8485262502beb51b0db716254

Any comment is really appreciated

mregen, 2017-12-11:
Hi @stefanocadoaleph, where did you get the yocto linux build with mono for the IOT 2000? I have a IOT2000 and could try to test/integrate your solution as we are keen to get mono/android working.

Cheers,
Martin

stefanocadoaleph, 2017-12-11:
Hi @mregen, You can cook by yourself an image by using this file https://github.com/aleph-solutions/UA-.NETStandard/commit/cef05bd7efa9e433f4b97b36cee84f1c93b01655 and by following [this guide](https://github.com/siemens/meta-iot2000) or If you prefer I can send you the image by wetransfer.

Cheers,
Stefano

mregen, 2017-12-11:
Hi @stefanocadoaleph, please send your image, it would save me a lot of time! 
Cheers,
Martin

stefanocadoaleph, 2017-12-18:
Hi @mregen, I apologise for the delay here is the image. https://we.tl/HBUFhN88Pz

Cheers,
Stefano

mregen, 2017-12-21:
Hi @stefanocadoaleph, by coincidence I had the IoT 2000 image as well with mono after a few rebuilds when you posted the link .. so this probem is solved. Thanks for your support.
Anyway I continued testing Mono 5.4 on Ubuntu for simplicity. So far I verified not calling dispose on RSA fixes the disposed key pair issue. It's not a great solution to rely on the finalizers to dispose the RSA objects but it seems to be the only option at this point to become compatible with Mono before the issue is fixed in their stack. I plan to have a fix ready before Christmas break. I still investigate an issue where the test app creates a new on every start...

mregen, 2017-12-22:
Hi @stefanocadoaleph, based on your changes and a few more I have now a branch(https://github.com/mregen/UA-.NetStandardLibrary/tree/mono) that starts a ref server on Yocto/iot2000, but connecting with secure sessions just times out, so clients cannot connect securely. Do you experience the same issue?

In addition X509Store doesn't load existing private keys, only directory store can be used.

stefanocadoaleph, 2018-01-05:
Hi @mregen, happy new year!
X509Store doesn't work to me too; I'm using directory store.
I had issues regarding security when the server wasn't able to save certificates.

You know the first time client tries connecting with security, server rejects its certificate or you need to setup server to autoaccept client certificate, did you check that?

stefanocadoaleph, 2018-01-05:
Hi @mregen, maybe you are getting timeout because iot2000 takes time to validate client certificate (poor CPU performance).
I just cloned your branch and tried. I had to increase connection timeout of my UAExpert from 10s to higher value (I put 60000ms) and then connection with security had worked.

Have you already tried this solution?

mregen, 2018-01-08:
@stefanocadoaleph thanks for trying this out! I will do some more testing today with increased timeout to see how other encryption algorithms are affected..

mregen, 2018-01-10:
Hi @stefanocadoaleph, works for me too. I'll prepare a PR which handles the dispose properly at runtime. Thanks for your contribution!

stefanocadoaleph, 2018-01-12:
Hi @mregen, thank you for the update!

mregen, 2018-01-23:
fixed #325

closed 2018-01-23 by mregen