OPCFoundation/UA-.NETStandard #225: Memory leaking in TcpMessageSocket

EddiG, 2017-08-10:
So, my first attempt for this issue was not good. Will try to change it :)
I have created the [repo](https://github.com/EddiG/issue-opc-memory-leak) for demonstrating the issue.

Below the explanation why I have suggested the #217 PR.

Firstly I figured out that the `IMessageSocketAsyncEventArgs` is not disposed of in [`OnConnectComplete`](https://github.com/OPCFoundation/UA-.NETStandardLibrary/blob/master/Stack/Opc.Ua.Core/Stack/Tcp/UaSCBinaryClientChannel.cs#L723) event handler. I have tried to fix it by calling `e.Dispose()` in that handler, but the memory still leaks.

I don't understand why it is not working because, if we look at [`TcpMessageSocketAsyncEventArgs`](https://github.com/OPCFoundation/UA-.NETStandardLibrary/blob/master/Stack/Opc.Ua.Core/Stack/Tcp/TcpMessageSocket.cs#L50), we see that it should dispose of the wrapped `m_args` and all should be fine. But it's not.

Next, I have split the logic in [`BeginConnect`](https://github.com/OPCFoundation/UA-.NETStandardLibrary/blob/master/Stack/Opc.Ua.Core/Stack/Tcp/TcpMessageSocket.cs#L245) because it is too complicated, that I mention in #218. I have removed `TcpMessageSocketAsyncEventArgs` from this method because it is not necessary for connection establishing and replaced it by `SocketAsyncEventArgs`. After that memory leaking has been stopped.

Further, I have decided that if [`TcpMessageSocketAsyncEventArgs`](https://github.com/OPCFoundation/UA-.NETStandardLibrary/blob/master/Stack/Opc.Ua.Core/Stack/Tcp/TcpMessageSocket.cs#L50) is not necessary here and need only for passing [`WriteOperation`](https://github.com/OPCFoundation/UA-.NETStandardLibrary/blob/master/Stack/Opc.Ua.Core/Stack/Tcp/UaSCBinaryClientChannel.cs#L725) through the `UserToken` I can try to dispose of [`SocketAsyncEventArgs`](https://github.com/OPCFoundation/UA-.NETStandardLibrary/pull/217/commits/58673554b43cf560db502b742d0515410199422d) in [event handler](https://github.com/OPCFoundation/UA-.NETStandardLibrary/blob/5067b94767effa558faa2a057923ae8f407e686a/Stack/Opc.Ua.Core/Stack/Tcp/TcpMessageSocket.cs#L307) (#217). Because on connection establishing step a `SocketAsyncEventArgs` not used for sending/receiving data and keeps alive `UserToken` when disposed, it is work again. But surely it is not a good solution. A good solution would be rethinking of  the [`TcpMessageSocketAsyncEventArgs`](https://github.com/OPCFoundation/UA-.NETStandardLibrary/blob/master/Stack/Opc.Ua.Core/Stack/Tcp/TcpMessageSocket.cs#L50) and [`BeginConnect`](https://github.com/OPCFoundation/UA-.NETStandardLibrary/blob/master/Stack/Opc.Ua.Core/Stack/Tcp/TcpMessageSocket.cs#L245) implementation because of #218.

EddiG, 2017-08-10:
> I have removed TcpMessageSocketAsyncEventArgs from this method because it is not necessary for connection establishing and replaced it by SocketAsyncEventArgs. After that memory leaking has been stopped.

I have prepared [commit](https://github.com/EddiG/UA-.NETStandardLibrary/commit/e88d3ab5e3cad861e6e8cd1bf078f214f09c73f0) that demonstrate what I talked about.

mregen, 2017-08-22:
Hi @EddiG thanks for pointing out the issue and your deep investigation. As you have noticed this is a complex piece of code and needs some time for thorough review and retesting.

closed 2017-09-29 by mregen