OPCFoundation/UA-.NETStandard #229: Server doesn't load correct application certififcate

K47S, 2017-08-16:
https://github.com/OPCFoundation/UA-.NETStandardLibrary/blob/af0dc88099e372a474c2411c630f6e5a76ebe190/Stack/Opc.Ua.Core/Security/Certificates/DirectoryCertificateStore.cs#L298

This line should be replaced with:

`
if (subjectName.Contains("=") || !certificate.Subject.Equals("CN=" + subjectName))
`

Server was first started with subjectName = "Example-Server" --> Example-Server cert is created.
Afterwards I start the Server with subjectName = "Example" --> Example-Server cert is loaded. I expected that I'll generate a new cert ("Example").

mregen, 2017-08-22:
@K47S good catch, but I think a simple compare is not enough, because there could be more entries in the cert subject. First extract the distinguished names list from the cert, then compare just the CN= entry.

mregen, 2017-12-13:
Testing fix...
https://github.com/mregen/UA-.NetStandardLibrary/commit/256ac8e5bd3a58b978792f1538d7dc5d457f55ad

closed 2018-01-10 by mregen