OPCFoundation/UA-.NETStandard #234: Clients get connection timeout under Core 2.0 server

goyzhang, 2017-08-22:
If the server is compiled under .NET CORE 2.0，it still runs without any error but clients are not able to build any connection. Shouldn't it be downward compatible？

viados, 2017-08-23:
@goyzhang Recognized same issue here!
Are there any plans to bring the library from .NET Standard 1.6 to .NET Standard 2.0?

WillooWisp, 2017-08-24:
.NET Core 2.0 has been stable for quite some time now and is released. A lot of users have already migrated, so please make your library work or at least give us some information for how to fix the timeout issue.

mregen, 2017-08-24:
@viados, @WillooWisp, @goyzhang 
Please try this branch: https://github.com/OPCFoundation/UA-.NETStandardLibrary/tree/proto_netstandard2.0
I found a few cases which had never been hit by old .Net frameworks wrt SendAsync/ReceiveAsync. 
Also I still get some yellow bubbles in the UWP apps for the Opc core libs, but they seem to work.

eoursel, 2017-08-25:
Tested on Fedora 26 with Netcore 2.0. Test ok with Server+Client compiled in Netcore 2.0. Good work, guys

WillooWisp, 2017-08-28:
I can also confirm that my initial testing looks good. I will do some more long running test this week probably. Good work, when will it end up in the dev branch?

mregen, 2017-08-29:
see PR https://github.com/OPCFoundation/UA-.NETStandardLibrary/pull/237. But we need to run CTT against .NetStandard 2.0 before we can merge..

WillooWisp, 2017-08-30:
On ARM it does not seem to work, since I get BadSecureChannelClosed exception when creating a session, any ideas?

'Opc.Ua.ServiceResultException: BadSecureChannelClosed
   at Opc.Ua.Bindings.ChannelAsyncOperation`1.End(Int32 timeout, Boolean throwOnError) in D:\Projects\Git\Angular\web-iot-starter\net\Gateway\Drivers\OpcFoundation.OpcUA\Opc.Ua.Core\Stack\Tcp\ChannelAsyncOperation.cs:line 172
   at Opc.Ua.Bindings.UaSCUaBinaryClientChannel.EndSendRequest(IAsyncResult result) in D:\Projects\Git\Angular\web-iot-starter\net\Gateway\Drivers\OpcFoundation.OpcUA\Opc.Ua.Core\Stack\Tcp\UaSCBinaryClientChannel.cs:line 303
   at Opc.Ua.Bindings.UaSCUaBinaryTransportChannel.EndSendRequest(IAsyncResult result) in D:\Projects\Git\Angular\web-iot-starter\net\Gateway\Drivers\OpcFoundation.OpcUA\Opc.Ua.Core\Stack\Tcp\UaSCBinaryTransportChannel.cs:line 334
   at Opc.Ua.Bindings.UaSCUaBinaryTransportChannel.SendRequest(IServiceRequest request) in D:\Projects\Git\Angular\web-iot-starter\net\Gateway\Drivers\OpcFoundation.OpcUA\Opc.Ua.Core\Stack\Tcp\UaSCBinaryTransportChannel.cs:line 290
   at Opc.Ua.DiscoveryClient.GetEndpoints(RequestHeader requestHeader, String endpointUrl, StringCollection localeIds, StringCollection profileUris, EndpointDescriptionCollection& endpoints) in D:\Projects\Git\Angular\web-iot-starter\net\Gateway\Drivers\OpcFoundation.OpcUA\Opc.Ua.Core\Stack\Generated\Opc.Ua.Client.cs:line 4821
   at Opc.Ua.DiscoveryClient.GetEndpoints(StringCollection profileUris) in D:\Projects\Git\Angular\web-iot-starter\net\Gateway\Drivers\OpcFoundation.OpcUA\Opc.Ua.Core\Stack\Client\DiscoveryClient.cs:line 68

mregen, 2017-08-30:
@WillooWisp Which ARM platform are you using?

mregen, 2017-09-15:
PR #237 was merged.

closed 2017-09-15 by mregen

WillooWisp, 2017-09-15:
Raspberry PI 2/3 and Revolution PI.