OPCFoundation/UA-.NETStandard #240: Cannot create session on ARM, Raspbian Pi (.NET Core 2.0)

WillooWisp, 2017-08-30:
The proto_netstandard2.0 branch works on .NET Core 2.0 x64, but on ARM it does not seem to work, since I get BadSecureChannelClosed exception when creating a session, any ideas?

'Opc.Ua.ServiceResultException: BadSecureChannelClosed
at Opc.Ua.Client.Session.d__72.MoveNext() in D:\Projects\Git\Angular\web-iot-starter\net\Gateway\Drivers\OpcFoundation.OpcUA\SDK\Opc.Ua.Client\Session.cs:line 810
--- End of stack trace from previous location where exception was thrown ---
at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
at OpcUa.OpcUaDriver.d__48.MoveNext() in D:\Projects\Git\Angular\web-iot-starter\net\Gateway\Drivers\OpcUaDriver\OpcUaDriver.cs:line 643
--- End of stack trace from previous location where exception was thrown ---
at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
at OpcUa.OpcUaDriver.d__40.MoveNext() in D:\Projects\Git\Angular\web-iot-starter\net\Gateway\Drivers\OpcUaDriver\OpcUaDriver.cs:line 412'.

'Opc.Ua.ServiceResultException: BadSecureChannelClosed
at Opc.Ua.Bindings.ChannelAsyncOperation`1.End(Int32 timeout, Boolean throwOnError) in D:\Projects\Git\Angular\web-iot-starter\net\Gateway\Drivers\OpcFoundation.OpcUA\Opc.Ua.Core\Stack\Tcp\ChannelAsyncOperation.cs:line 172
at Opc.Ua.Bindings.UaSCUaBinaryClientChannel.EndSendRequest(IAsyncResult result) in D:\Projects\Git\Angular\web-iot-starter\net\Gateway\Drivers\OpcFoundation.OpcUA\Opc.Ua.Core\Stack\Tcp\UaSCBinaryClientChannel.cs:line 303
at Opc.Ua.Bindings.UaSCUaBinaryTransportChannel.EndSendRequest(IAsyncResult result) in D:\Projects\Git\Angular\web-iot-starter\net\Gateway\Drivers\OpcFoundation.OpcUA\Opc.Ua.Core\Stack\Tcp\UaSCBinaryTransportChannel.cs:line 334
at Opc.Ua.Bindings.UaSCUaBinaryTransportChannel.SendRequest(IServiceRequest request) in D:\Projects\Git\Angular\web-iot-starter\net\Gateway\Drivers\OpcFoundation.OpcUA\Opc.Ua.Core\Stack\Tcp\UaSCBinaryTransportChannel.cs:line 290
at Opc.Ua.DiscoveryClient.GetEndpoints(RequestHeader requestHeader, String endpointUrl, StringCollection localeIds, StringCollection profileUris, EndpointDescriptionCollection& endpoints) in D:\Projects\Git\Angular\web-iot-starter\net\Gateway\Drivers\OpcFoundation.OpcUA\Opc.Ua.Core\Stack\Generated\Opc.Ua.Client.cs:line 4821
at Opc.Ua.DiscoveryClient.GetEndpoints(StringCollection profileUris) in D:\Projects\Git\Angular\web-iot-starter\net\Gateway\Drivers\OpcFoundation.OpcUA\Opc.Ua.Core\Stack\Client\DiscoveryClient.cs:line 68

eoursel, 2017-08-30:
Looks like the BadSecureChannelClosed is sent when the the certficate of the clients is not installed on the server. Did you tried a connection without security? 

What version of Raspbian are you testing? I am interested to do some tests myself next week.

WillooWisp, 2017-09-07:
I have tried against the same OPC UA server with same settings as when it works on Windows 10, x64. I use the distributions from https://packages.revolutionpi.de/dists/, since testing the client on the industrial RevPi.

Looking forward to hearing from your test results.

WillooWisp, 2017-09-07:
Anyone else tried this on an ARM CPU of any kind?

mregen, 2017-09-11:
@WillooWisp The server side seems to be working: I installed Raspbian according to these instructions: 
https://blogs.msdn.microsoft.com/david/2017/07/20/setting_up_raspian_and_dotnet_core_2_0_on_a_raspberry_pi/
Then I published the ConsoleReferenceServer (latest .Net2.0 enabled branch) on a USB stick and started it on the pi. I can connect to the server from my windows box.

mregen, 2017-10-26:
@WillooWisp Hi, is the issue resolved?

closed 2017-10-27 by mregen

WillooWisp, 2017-10-28:
It is indeed!