OPCFoundation/UA-.NETStandard #242: GeneralModelChangeEventType event Subscription in .NET Core UA Client does not notify for model change.

JayeshThamke, 2017-09-10:
Hello,
I made following code changes in NetCoreConsoleClient app, because I want to subscribe GeneralModelChangeEventType to observe change in UA Server Information Model. 
          `var subscription = new Subscription(session.DefaultSubscription) { PublishingInterval = 1000 };

            Console.WriteLine("6 - Add a list of items (server current time and status) to the subscription.");
            var list = new List<MonitoredItem> {
                new MonitoredItem(subscription.DefaultItem)
                {
                    DisplayName = "ServerStatusCurrentTime", StartNodeId = "i=2258"
                },

                new MonitoredItem(subscription.DefaultItem, true) // new code - Monitored Item for event 
                {
                    DisplayName = "ModelChangedEvent",
                    StartNodeId = ObjectTypeIds.GeneralModelChangeEventType
                } // new code - ends
            };

            list.ForEach(i => i.Notification += OnNotification);

            subscription.AddItems(list);`

By manually creating and deleting a dynamic variable node using create and delete node methods, however it does not notify for model changes in OnNotification handler method, see below.

           `private static void OnNotification(MonitoredItem item, MonitoredItemNotificationEventArgs e)
             {
            if (item.DisplayName == "ModelChangedEvent")
            { // breakpoint 

            }

            foreach (var ev in item.DequeueEvents())
            {
                Console.WriteLine(ev.Message);
            }

            foreach (var value in item.DequeueValues())
            {
                Console.WriteLine("{0}: {1}, {2}, {3}", item.DisplayName, value.Value, value.SourceTimestamp, value.StatusCode);
            }

        }`

Is my implementation correct or any other way to implement, assistance is appreciated.
**Note:** Updating Server Information Model in UaExpert ua client works fine.

Thanks ,
Jay

AlinMoldovean, 2017-09-11:
Hello @JayeshThamke ,

For subscribing to notifications for event monitored items you need to create a subscription and a monitored item for the "EventNotifier" attribute of the Server node (ns=0;i=2253).
In your sample code you a re creating a monitored item for the "Value" attribute of the event type node.

JayeshThamke, 2017-09-12:
Hallo @AlinMoldovean 
**1. About EventNotifier Attribute:**
The Object Node _Server_ by-default has _EventNotifier_ Node Attribute set to _SubscribeToEvents_ bit.

To be very precise, the dynamic nodes appear/disappear under Object Node (FolderType) _Demo_ . The Ua Server has an Object Node called _Demo_, its _EventNotifier_ Attribute is set to None as shown in fig below.

I would like to know whether I have to set _EventNotifier_ to _SubscribeToEvents_ bit for _Demo_ Object Node beforehand? Thank you

**2. About creating monitored Items:**
As per your suggestion, when creating MonitoredItem for subscription, are setting up following MonitoredItem attributes enough?  Thank you

                `new MonitoredItem(subscription.DefaultItem, true)
                {
                                    AttributeId = Attributes.EventNotifier,
                                    DisplayName = "Demo",
                                    StartNodeId = "ToBeSearchedCorrectNodeId"
                 }`

[image]

AlinMoldovean, 2017-09-17:
Hello @JayeshThamke ,
The event monitored item should be created for the standard ""Server" node.
The "Demo" node of your server will not generate ModelChange events by default, even if you set it's event notifier attribute.

JayeshThamke, 2017-09-17:
Hello @AlinMoldovean 
Thank you for reply, 
Is there any way to detect dynamic nodes under Demo (FolderType) Object Node?
Thanks,
Jayesh

JayeshThamke, 2017-09-17:
Hi @AlinMoldovean,
My question is more specific towards implementation of **auto-browse** feature in client application, whenever the new (Instance) Nodes appears or disappears under Demo node. How to achieve it? 
Thanks for suggestions,
Jayesh T

JayeshThamke, 2017-09-27:
Hello @AlinMoldovean 
I already have a NodeVersion Variable Node which changes its Value attribute as the nodes are created or removed dynamically from Server's Information Model under Demo node shown in an image above. 
I wonder if it is the correct way to proceed and detect changes in Information model. 
Thanks,
Jay

closed 2018-02-06 by mregen