OPCFoundation/UA-.NETStandard #243: Complex Data Types from Server to Client

kdb2009, 2017-09-12:
Hello all,

struggeling to get complex data types known by the Server into my Client. Also I need them for several method calls e.g. calling "ScanStart" with a paramater from the specific type "ScanSettings" which is only provided by the server. Is it possible to get such a (struct) data type with default values in a simple way?

Does somebody know how and can provide an example?

Best regards

awcullen, 2017-09-24:
Below is an sample that calls a method with argument of type "Vector", which is a custom structured type that is provided by UnifiedAutomation's demo server "UaCppServer". I first used UaExpert to discover the DataTypeId, EncodingId, ObjectId, MethodId,  as well as guess the structure.

```
using System;
using System.Collections.Generic;
using System.Threading.Tasks;
using Opc.Ua;
using Opc.Ua.Client;
using System.Security.Cryptography.X509Certificates;

namespace NetCoreConsoleClient
{
    public class Program
    {
        public static void Main(string[] args)
        {
            Console.WriteLine(".Net Core OPC UA Console Client sample");
            string endpointURL;
            if (args.Length == 0)
            {
                // use OPC UA .Net Sample server 
                endpointURL = "opc.tcp://" + Utils.GetHostName() + ":48010"; // UaCPPServer
            }
            else
            {
                endpointURL = args[0];
            }
            try
            {
                Task t = ConsoleSampleClient(endpointURL);
                t.Wait();
            }
            catch (Exception e)
            {
                Console.WriteLine("Exit due to Exception: {0}", e.Message);
            }
        }

        public static async Task ConsoleSampleClient(string endpointURL)
        {
            Console.WriteLine("1 - Create an Application Configuration.");
            Utils.SetTraceOutput(Utils.TraceOutput.DebugAndFile);
            var config = new ApplicationConfiguration()
            {
                ApplicationName = "UA Core Sample Client",
                ApplicationType = ApplicationType.Client,
                ApplicationUri = "urn:" + Utils.GetHostName() + ":OPCFoundation:CoreSampleClient",
                SecurityConfiguration = new SecurityConfiguration
                {
                    ApplicationCertificate = new CertificateIdentifier
                    {
                        StoreType = "X509Store",
                        StorePath = "CurrentUser\\UA_MachineDefault",
                        SubjectName = "UA Core Sample Client"
                    },
                    TrustedPeerCertificates = new CertificateTrustList
                    {
                        StoreType = "Directory",
                        StorePath = "OPC Foundation/CertificateStores/UA Applications",
                    },
                    TrustedIssuerCertificates = new CertificateTrustList
                    {
                        StoreType = "Directory",
                        StorePath = "OPC Foundation/CertificateStores/UA Certificate Authorities",
                    },
                    RejectedCertificateStore = new CertificateTrustList
                    {
                        StoreType = "Directory",
                        StorePath = "OPC Foundation/CertificateStores/RejectedCertificates",
                    },
[... 93 lines cut ...]
            public override ExpandedNodeId TypeId => ExpandedNodeId.Parse("nsu=http://www.unifiedautomation.com/DemoServer/;i=3002");

            public override ExpandedNodeId BinaryEncodingId => ExpandedNodeId.Parse("nsu=http://www.unifiedautomation.com/DemoServer/;i=5054");

            public override ExpandedNodeId XmlEncodingId => NodeId.Null;

            public override void Encode(IEncoder encoder)
            {
                encoder.WriteDouble("X", this.X);
                encoder.WriteDouble("Y", this.Y);
                encoder.WriteDouble("Z", this.Z);
            }

            public override void Decode(IDecoder decoder)
            {
                this.X = decoder.ReadDouble("X");
                this.Y = decoder.ReadDouble("Y");
                this.Z = decoder.ReadDouble("Z");
            }

            public override string ToString() => $"{{ X={this.X}; Y={this.Y}; Z={this.Z}; }}";
        }
    }
}
```

AlinMoldovean, 2017-09-27:
Hello @kdb2009,

Before being able to do this the client should fetch the custom structured data types and enumerated types known by the server.
They are exposed in some variable node under the “Types\DataTypes\OPC Binary” node.

Those nodes should have a ByetSting value containing a binary encoded TypeDictionary object.
The client should parse these nodes and read the list of structured data types from the dictionary.
Some further browse and read operations are required to get some other information about each data type, like EncodingId(s) or DataTypeId.
Using that information the client will be able to automatically encode or decode instances of complex data types.
For the moment this feature is provided only in some commercial OPC UA SDKs, e.g from Softing or Unified Automation.

In the sample code from @awcullen the client already has some preconfigured knowledge about the content of a "Vector" type.

[image]

<opc:TypeDictionary
  xmlns:opc="http://opcfoundation.org/BinarySchema/"
  xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
  xmlns:ua="http://opcfoundation.org/UA/"
  xmlns:tns="http://industrial.softing.com/UA/Refrigerator"
  DefaultByteOrder="LittleEndian"
  TargetNamespace="http://industrial.softing.com/UA/Refrigerator"
>  
  <opc:Import Namespace="http://opcfoundation.org/UA/" Location="Softing.Opc.Ua.Sdk.BinarySchema.bsd"/>

  <opc:StructuredType Name="RefrigeratorStatusType" BaseType="ua:ExtensionObject">
    <opc:Field Name="CondensorMotorRunning" TypeName="opc:Boolean" />
    <opc:Field Name="PreasureBeforePump" TypeName="opc:Double" />
    <opc:Field Name="PreasureAfterPump" TypeName="opc:Double" />
  </opc:StructuredType>

  <opc:EnumeratedType Name="EnumRefrigeratorState" LengthInBits="32">
    <opc:EnumeratedValue Name="Stopped" Value="0" />
    <opc:EnumeratedValue Name="Running" Value="1" />
    <opc:EnumeratedValue Name="Fault" Value="2" />
  </opc:EnumeratedType>

</opc:TypeDictionary>

kdb2009, 2017-09-29:
Hello @AlinMoldovean,

First thank you for your detailed answer.

So the example below contains two examples for enumerated types and structured types? 
Do I have to rebuild this xml-file on my own?
And if I have to do this, what is the next step? Where does the program except this xml-file?

Best Regards

AlinMoldovean, 2017-10-03:
Hello @kdb2009,

Yes, the dictionary from my example contains the definition for a structured type and an enumerated type.
But the client stack needs some further implementation before being able to handle these custom types. And this feature is missing for the moment.
The Encoder/Decoder classes (BinaryEncoder and BinaryDecoder at least) need to be extended for this.
An option is to implement a generic class for handling all structured type instances.
Something like this:

[image]

closed 2018-04-19 by AlinMoldovean

Abarbz, 2018-05-14:
Thanks @AlinMoldovean  for pointing me to this issue.
I don't understand your second solution though (I would really like to avoid having to map my complex struct statically on the client side like in the solution by awcullen). 
Would you mind sharing a snippet of code?

Noires, 2018-05-15:
Iam currently implementing it with a abstract type of IEncodeable, where i create at runtime new types via Reflection and Emit based on the parsed data of the type dictionary. The only thing iam currently lacking is the ExpandableNodeId so i can add the type properly to the EncodeableFactory and then everything should work. Can the client get it somehow?

mregen, 2019-10-31:
see also #838