OPCFoundation/UA-.NETStandard #251: DOT NET CORE 2.0 Compatibility

chandru9050, 2017-09-25:
Hi ,
                I have installed dotnetcore2.0 in my machine , While compiling the core library I am getting this below error 

>Opc.Ua.Core -> C:\UA-.NETStandardLibrary-master (2)\UA-.NETStandardLibrary-master\Stack\Opc.Ua.Core\bin\Debug\netstandard1.4\Opc.Ua.Core.dll
1>Stack\Bindings\BaseBinding.cs(24,41,24,48): error CS0012: The type 'Object' is defined in an assembly that is not referenced. You must add a reference to assembly 'netstandard, Version=2.0.0.0, Culture=neutral, PublicKeyToken=cc7b13ffcd2ddd51'.
1>Stack\Bindings\BaseBinding.cs(50,16,50,37): error CS0012: The type 'Object' is defined in an assembly that is not referenced. You must add a reference to assembly 'netstandard, Version=2.0.0.0, Culture=neutral, PublicKeyToken=cc7b13ffcd2ddd51'.
1>Stack\Bindings\BaseBinding.cs(31,13,31,27): error CS0012: The type 'Object' is defined in an assembly that is not referenced. You must add a reference to assembly 'netstandard, Version=2.0.0.0, Culture=neutral, PublicKeyToken=cc7b13ffcd2ddd51'.
1>Stack\Bindings\BaseBinding.cs(32,13,32,30): error CS0012: The type 'Object' is defined in an assembly that is not referenced. You must add a reference to assembly 'netstandard, Version=2.0.0.0, Culture=neutral, PublicKeyToken=cc7b13ffcd2ddd51'.
1>Stack\Bindings\BaseBinding.cs(33,13,33,34): error CS0012: The type 'Object' is defined in an assembly that is not referenced. You must add a reference to assembly 'netstandard, Version=2.0.0.0, Culture=neutral, PublicKeyToken=cc7b13ffcd2ddd51'.
1>Stack\Bindings\BaseBinding.cs(58,17,58,38): error CS0012: The type 'Object' is defined in an assembly that is not referenced. You must add a reference to assembly 'netstandard, Version=2.0.0.0, Culture=neutral, PublicKeyToken=cc7b13ffcd2ddd51'.
1>Types\Schemas\OPCBinarySchema.cs(345,31,345,50): error CS0234: The type or namespace name 'XmlIncludeAttributeAttribute' does not exist in the namespace 'System.Xml.Serialization' (are you missing an assembly reference?)
1>Types\Schemas\OPCBinarySchema.cs(345,31,345,50): error CS0234: The type or namespace name 'XmlIncludeAttribute' does not exist in the namespace 'System.Xml.Serialization' (are you missing an assembly reference?)
1>Types\Schemas\OPCBinarySchema.cs(346,31,346,50): error CS0234: The type or namespace name 'XmlIncludeAttributeAttribute' does not exist in the namespace 'System.Xml.Serialization' (are you missing an assembly reference?)
1>Types\Schemas\OPCBinarySchema.cs(346,31,346,50): error CS0234: The type or namespace name 'XmlIncludeAttribute' does not exist in the namespace 'System.Xml.Serialization' (are you missing an assembly reference?)
1>Types\Schemas\OPCBinarySchema.cs(347,31,347,50): error CS0234: The type or namespace name 'XmlIncludeAttributeAttribute' does not exist in the namespace 'System.Xml.Serialization' (are you missing an assembly reference?)
1>Types\Schemas\OPCBinarySchema.cs(347,31,347,50): error CS0234: The type or namespace name 'XmlIncludeAttribute' does not exist in the namespace 'System.Xml.Serialization' (are you missing an assembly reference?)
1>Types\Schemas\OPCBinarySchema.cs(351,31,351,47): error CS0234: The type or namespace name 'XmlTypeAttributeAttribute' does not exist in the namespace 'System.Xml.Serialization' (are you missing an assembly reference?)
1>Types\Schemas\OPCBinarySchema.cs(351,31,351,47): error CS0234: The type or namespace name 'XmlTypeAttribute' does not exist in the namespace 'System.Xml.Serialization' (are you missing an assembly reference?)
1>Types\Schemas\OPCBinarySchema.cs(351,48,351,57): error CS0246: The type or namespace name 'Namespace' could not be found (are you missing a using directive or an assembly reference?)
1>Types\Schemas\BinarySchema.cs(26,10,26,19): error CS0246: The type or namespace name 'XmlIgnor

mregen, 2017-09-27:
@chandru9050 Please provide more context: OS, VS version, How do you start the build command? Which Solution? Make sure you have the latest VS 2017 updates and .Net Core 2.0 SDK.

chandru9050, 2017-09-28:
Thank you regen for your response, I figured out the problem, I updated my Visual studio 2017  with  Version 15.3.5 , then things started working like magic. Thanks again

closed 2017-09-28 by mregen