OPCFoundation/UA-.NETStandard #252: Issue with complex events on MonitoredItems

Noires, 2017-09-25:
I'am creating my own address space with the model compiler and having there complex events defined by a specification and using the EventView in **UaExpert** to verify if i can subscribe the events correctly.

The only issue i got with the events are events which contain objects (like a section) and those contain objects (like a subsection). The way it would work would define all of those objects as **Mandatory**, then i could monitor those properties. The problem here is then i would see all sections and subsections which are actually optional. 

If i use either **Optional** or **OptionalPlaceholder** than **Mandatory**, they are not listed under the event type in the UaExpert Eventview. 

Any idea what i should change? At the moment i'am unsure if it is just not intended, if the fault lies in the UaExpert or in the UA-.NET StandardLibrary. How does the client knows which parameters from a event he can subscribe? Where might i be able to check that?

Greetings 
Noires

AlinMoldovean, 2018-05-03:
closed due to inactivity

closed 2018-05-03 by AlinMoldovean

mregen, 2019-10-31:
see #838