OPCFoundation/UA-.NETStandard #261: Configuration file can't load

Pedersen-dev, 2017-10-01:
Hi,

Sitting on a macbook air, sierra version 10.12.6.

I'm getting following messages each time i try to run NetCoreConsoleServer as described in the Readme guide
[image]
 about NetCoreConsoleClient and NetCoreConsoleServer:
_Load Application Configuration: The platform does not have a definition for an X509 certificate store named 'UA_MachineDefault' with a StoreLocation of 'CurrentUser', and does not support creating it.
Exception: One or more errors occurred. (Could not load configuration file.)_

Furthermore it does not create the OPC Foundation\CertificateStores\RejectedCertificates and OPC Foundation\CertificateStores\UA Applications folders.

Anyone have a solution for this or know what i am doing wrong ? 

(i have no problem to get it to run with ubuntu and windows)

mregen, 2017-10-02:
@Pedersen-dev Thanks for the reporting the issue, looks as if there is a problem with the X509 cert store on macOS.

As a workaround you can change the following lines in Opc.Ua.SampleServer.Config.xml to:

`<ApplicationCertificate>`
`<StoreType>Directory</StoreType>`
`<StorePath>OPC Foundation/CertificateStores/MachineDefault</StorePath>`
`<SubjectName>CN=UA Core Sample Server, C=US, S=Arizona, O=OPC Foundation, DC=localhost</SubjectName>`
`</ApplicationCertificate>`

The change will use the Directory to store the application certs.

Pedersen-dev, 2017-10-02:
Thank you. It is working for the server now.

Next problem is the exact same problem just for client:
[image]

I already try'ed to apply the same change to Opc.Ua.Sampleclient.Config.xml as i did for Opc.Ua.SampleServer.Config.xml without any luck.

Any suggestion ?

mregen, 2017-10-04:
Hi @Pedersen-dev , the console client doesn't use the configuration library.
Please change the code in Program.cs:

                     ApplicationCertificate = new CertificateIdentifier
                    {
                        StoreType = "X509Store",
                        StorePath = "CurrentUser\\UA_MachineDefault",
                        SubjectName = "UA Core Sample Client"
                    },

to 

                        StoreType = "Directory",
                        StorePath = "OPC Foundation/CertificateStores/MachineDefault",

mregen, 2017-10-04:
This issue was also reported here: https://github.com/dotnet/corefx/issues/19508.

In .NET Core 1.0 and .NET Core 1.1 the macOS implementation of the cryptography classes was based on OpenSSL. In .NET Core 2.0 the dependency was changed to use Apple's Security.framework. I need to investigate a fix for X509Store which works on all platforms.

mregen, 2017-10-05:
@Pedersen-dev I tested on a Mac and another workaround is to use the X509Store CurrentUser\My.

mregen, 2017-10-12:
fixed

closed 2017-10-12 by mregen