OPCFoundation/UA-.NETStandard #266: Session is created even after self signed certificate is rejected

JayeshThamke, 2017-10-09:
Hello,
I am developing a OPC UA Client, and in `CertificationValidationHandler`  method, if I rejected the OPC UA Server's certificate the Session can still be created between my OPC UA Client and the OPC UA Server. 
I can see the rejected certificate in RejectedCertStore (repository) 

I wonder if it is the correct behavior.

Following is the code from CertificationValidationHandler method for cert rejection and adding it into RejectedCertStore.

`e.Accept = false;
var untrustedCertStore = Configuration.SecurityConfiguration.RejectedCertificateStore.OpenStore();
X509Certificate2 untrustedCert = new X509Certificate2(e.Certificate.RawData);
try
{
var untrustedCertCollection = untrustedCertStore.FindByThumbprint(untrustedCert.Thumbprint);
   if (untrustedCertCollection.Result.Contains(e.Certificate) == false)
       {
          untrustedCertStore.Add(e.Certificate).Wait();
        }
 }
   finally
   {
      untrustedCertStore.Close();
   }`

Please correct me if I am wrong.

Thanks in advance for your reply,
Jayesh

mregen, 2017-10-10:
Hi @JayeshThamke, the certificate in the rejected store is only a copy for the records and not a negative list .. e.g. an admin can copy a rejected cert to the trusted store to trust it. It doesn't mean it is rejected forever. If somebody tries again with the same cert it can be approved by the validation handler too, even if it is already in the rejected store.

Cheers,
Martin

closed 2017-10-26 by mregen

banshandeqie, 2023-12-06:
> Hi @JayeshThamke, the certificate in the rejected store is only a copy for the records and not a negative list .. e.g. an admin can copy a rejected cert to the trusted store to trust it. It doesn't [...]

Can you give a code example? Look at the code do not know how to deal with.

mregen, 2023-12-06:
Hi @banshandeqie , please open a new issue with the specific question you have. Your comment is otherwise easily overseen.