OPCFoundation/UA-.NETStandard #271: SDK available Documentation

fcl-93, 2017-10-17:
Hello

I'm trying to use this SDK to develop a client, and I want to know if there is any extra 
documentation besides the Samples that you have and the Wiki? Do you have any Reference guide?  I'm using an old version of a .chm file as a guide (Rel. 1.00.250.0). 

Another question is related with the service sets support is there any way to know which services are supported by the SDK?

I don't know if this is the right place to put this type of question, please if you have some forum or blog just let me know.

Thanks in advance
BR Fábio

mregen, 2017-10-26:
@fcl-93 Samples is all we have. You can also check the .Net repo for more samples which were not ported (yet).

Service set? You mean DA/HA etc?

Thanks,
Martin

fcl-93, 2017-10-26:
Hi @mregen 
When I talk about services sets I’m referring to Part 4 of OPC UA Specification. They have Query Service Set which are composed by QueryFirst Service, and QueryNext Service, or to Node Management Service Set that has AddNodes Service, AddReferences Service, DeleteNodes Service, DeleteReferences Service, etc… (**I just wanted to know which ones are fully implemented**)
**Do you think this SDK is stable enough to develop a client or it is better to search for commercial solutions to develop it?**
Thank for you answer 😊.
Fábio

barnstee, 2017-10-27:
The answer to your question is, as many times, "that depends" :-). In general, the OPC Foundation recommends to reach out to a commercial stack vendor for commercial applications. The SDK here is a reference implementation, without technical support. This support is also available by commercial stack vendors. So if you are building a commercial product, it is best to reach out to the stack vendors (Unified Automation, Softing, ProSysOPC, etc.) instead of integrating this as is. If, on the other hand, "you know what you're doing" or you are building a PoC, you can use this stack.

fcl-93, 2017-10-27:
Thank you for your help 👍

closed 2017-10-27 by fcl-93