OPCFoundation/UA-.NETStandard #272: ServiceResultException : SHA1 signed certificates are not trusted

geotinc, 2017-10-17:
When I run the NetCoreConsoleClient sample,

I have the following exception on the Session.Create call:

Opc.Ua.ServiceResultException : 
ServiceResultException : SHA1 signed certificates are not trusted

Can anybody help me?

Thanks

ThomasLevering, 2017-10-18:
SecurityConfiguration = new SecurityConfiguration
...
RejectSHA1SignedCertificates = false

geotinc, 2017-10-18:
I have after the following exception:
ServiceResultException : Certificate doesn't meet minimum key length requirement
So I have added MinimumCertificateKeySize = 64 In securityConfiguration

But I have now:
ServiceResultException : Certificate is not trusted.
SubjectName: CN=KEPServerEX/UA Server, O=Unknown, C=FR, DC=L-CND6090HB9
IssuerName: CN=KEPServerEX/UA Server, O=Unknown, C=FR, DC=L-CND6090HB9

I try to change AutoAcceptUntrustedCertificates to true without success

barnstee, 2017-10-18:
AutoAcceptUntrustedCertificates is a server setting. You need to manually trust the cert in the certificate validation callback of the client.

geotinc, 2017-10-18:
I have another exception (hope it is the last sorry...)
I try to connect to kepware (see endpoints bellow):
[image]

The exception is:
Error received from remote host

It occur at 
session.Open(sessionName, sessionTimeout, identity, preferredLocales, checkDomain);
Line 805 of Session.cs file

Thanks a lot

mregen, 2017-10-18:
@geotinc Regarding: ServiceResultException : Certificate is not trusted.
It is expected that a client or server don't connect with security until the certificates are trusted. The core client has a rejected folder under OPC Foundation in the project folder --> see also . Move the rejected cert to 'UA Applications' to establish trust from the client side. Your server may also need a similar procedure to establish trust to your server.

The kepware server also needs to trust the client, not sure about the procedure but the dialog box above indicates a 'trusted clients' tab, and the cert of the client must be added to it for trust.

geotinc, 2017-10-18:
Thanks a lot, everything works great now

[image]

closed 2017-10-18 by geotinc