OPCFoundation/UA-.NETStandard #273: TypeId from ReferenceDescription

geotinc, 2017-10-20:
I try to get the value datatype of a node Id without success all datatype is Numeric, cf the following code

               `session.Browse(null, null, ExpandedNodeId.ToNodeId(nodeId, session.NamespaceUris),
                0u, BrowseDirection.Forward, ReferenceTypeIds.HierarchicalReferences,
                true, (uint)NodeClass.Variable | (uint)NodeClass.Object | (uint)NodeClass.Method,
                out _, out ReferenceDescriptionCollection nextRefs);

            foreach (ReferenceDescription rd in nextRefs)
            {
                if (rd.NodeClass == NodeClass.Variable)
                {
                    ServerVariable serverVariable = new ServerVariable();
                    serverVariable.NodeId = ExpandedNodeId.ToNodeId(rd.NodeId, session.NamespaceUris);
                    serverVariable.IdType = rd.TypeDefinition.IdType;
                    serverVariable.Name = parentServerObject.Name + "." + rd.DisplayName;
                    parentServerObject.Variables.Add(serverVariable);
                    continue;
                }`

But the line 'serverVariable.IdType = rd.TypeDefinition.IdType;' always return Numeric, not Boolean, String, Array...

awcullen, 2017-10-20:
There are too many attributes named 'Type' in this specification!

If you are looking for the data type of the _value_ of a node of class Variable, you will need to follow these steps;
1. Get the NodeId of the Variable (as you have done).
2. Read the two attributes, DataType and ValueRank of the Node. DataType is another NodeId
3. Use the Opc.Ua.TypeInfo class to get the .NET system type. 
```
var builtInType = GetBuiltInType(datatypeId)
var sysType = GetSystemType(builtInType, valueRank);
```

geotinc, 2017-10-20:
Thanks but I don't find how to read attributes:

`

List<ReadValueId> nodesToRead = new List<ReadValueId>();
nodesToRead.Add(new ReadValueId { NodeId = serverVariable.NodeId, AttributeId = Attributes.ValueRank });
nodesToRead.Add(new ReadValueId { NodeId = serverVariable.NodeId, AttributeId = Attributes.DataType });`

Don't find which arguments must I give to session.Read method.
Thanks

geotinc, 2017-10-20:
Ok with the following code
[image]

closed 2017-10-20 by geotinc

awcullen, 2017-10-20:
Something like:
```
var nodesToRead = new [] 
{
new ReadValueId { NodeId = serverVariable.NodeId, AttributeId = Attributes.ValueRank},
new ReadValueId { NodeId = serverVariable.NodeId, AttributeId = Attributes.DataType}
};
session.Read(null, 0.0, TimestampsToReturn.None, nodesToRead, out results, out diagnosticInfos);

```