OPCFoundation/UA-.NETStandard #278: GDS is still an NET 4.6 app

eoursel, 2017-11-04:
Is anyone working on the GDS stuff? I noticed that the GDS server is still a .NET 4.6 target binary. It would be nice to have a NETCORE version since we are in the NETStandardLibrary and because it could run in a docker container since SQLServer 2017 is now available as a docker container (thanks Microsoft). 

Second point, this issue is already existing, but i should mention that an implementatin of the PUSH model inside a NetCoreConsolerServer would be very helpful. I believe even in the .NET we don't have any example of a server implementing the PUSH model. Am i wrong? 

Let me know if you need some help or you have a roadmap. 

Thanks.

mregen, 2017-11-06:
@eoursel Hi, I'm working on the GDS and push support. You can take a look on my working branch https://github.com/mregen/UA-.NETStandardLibrary/tree/gds_work_certs if you are interested. GDS client/server libs are now built for .Net Standard. Next PR should be ready by next week.
What kind of help could you provide?

closed 2017-12-08 by mregen

mregen, 2017-12-11:
docker sample #297