OPCFoundation/UA-.NETStandard #282: Get all available nodes from server (performance issue)

CodingHarry, 2017-11-13:
Are there other options to get all datapoint nodes than with `Session.Browse()` recursive.

We have 9 servers within our infrastructure and want to cache all available nodes for performance reasons and for search capabilities. So our application clients will just consume this cache.

At initialization time we browse all servers to get about 120.000 nodes from each of them. This takes about 3 up to 5 minutes on each server.

So the question if there is a more efficient way to get all nodes?

AlinMoldovean, 2017-11-16:
Hello @CodingHarry ,

For the moment there is no standard procedure in OPC UA (e.g Service call) to retrieve the complete content of the Server's Address Space on the client side.
This is because the Address Space of the Server is dynamic by design and it can modify consistently during the runtime.
Using a recursive Browse can be an option, but it can create performance issues on both client and server side.
I can suggest to have the server nodes (or a subset) saved on the client side as NodeSet2.xml files and use them as a local cache in case your client has some preconfigured knowledge about the server.
You can create the NodeSet2.xml files during a client session or have them from the server admistrator.

Thanks!

gava88, 2017-11-20:
Hello @AlinMoldovean,
I'm trying to save a subset of the server nodes and load them on the client side as NodeSet2.Xml as you suggested.

These are the server save methods:
```
NodeStateCollection nodesToSave = new NodeStateCollection();

BaseObjectState test = new BaseObjectState(null);
test.Create(SystemContext, new NodeId("TestObject", m_namespaceIndex), new QualifiedName("TestObject", m_namespaceIndex), null, false);
nodesToSave.Add(test);
BaseObjectState test2 = new BaseObjectState(null);
test2.Create(SystemContext, new NodeId("TestObject2", m_namespaceIndex), new QualifiedName("TestObject2", m_namespaceIndex), null, false);
nodesToSave.Add(test2);       
BaseDataVariableState variable = new BaseDataVariableState(null);
variable.Create(SystemContext, new NodeId("MyVar", m_namespaceIndex), new QualifiedName("MyVar", m_namespaceIndex), null, true);
variable.DataType = DataTypes.Float;
variable.Value = 19.5;
nodesToSave.Add(variable);

using (Stream stream = new FileStream("Programming.NodeSet2.xml", FileMode.Create))
{
           nodesToSave.SaveAsNodeSet2(SystemContext, stream);
}
using (Stream stream = new FileStream("Programming.NodeSet.xml", FileMode.Create))
{
           nodesToSave.SaveAsNodeSet(SystemContext, stream);
}
using (Stream stream = new FileStream("Programming.xml", FileMode.Create))
{
           nodesToSave.SaveAsXml(SystemContext, stream);
}
using (Stream stream = new FileStream("Programming.uanodes", FileMode.Create))
{
         nodesToSave.SaveAsBinary(SystemContext, stream);
}
```
Client load method:
```
            NodeStateCollection nodes = new NodeStateCollection();
            using (Stream stream = new FileStream("Programming.NodeSet2.xml", FileMode.Open))
            {
                try
                {
                    nodes.LoadFromBinary(session.SystemContext, stream, true);
                }
                catch (Exception e)
                {
                    Console.WriteLine(e.Message);
                }
            }
```
            

When I try to execute load method on the client, I get this exception:

> {Opc.Ua.ServiceResultException: MaxStringLength 65535 < 1819113535
>    at Opc.Ua.BinaryDecoder.ReadString(String fieldName, Int32 maxStringLength) in C:\Users\gavagnaf\Desktop\UA\UA2\UA-.NETStandardLibrary-master\UA-.NETStandardLibrary-master\Stack\Opc.Ua.Core\Types\Encoders\BinaryDecoder.cs:line 416
>    at Opc.Ua.BinaryDecoder.ReadString(String fieldName) in C:\Users\gavagnaf\Desktop\UA\UA2\UA-.NETStandardLibrary-master\UA-.NETStandardLibrary-master\Stack\Opc.Ua.Core\Types\Encoders\BinaryDecoder.cs:line 392
>    at Opc.Ua.BinaryDecoder.LoadStringTable(StringTable stringTable) in C:\Users\gavagnaf\Desktop\UA\UA2\UA-.NETStandardLibrary-master\UA-.NETStandardLibrary-master\Stack\Opc.Ua.Core\Types\Encoders\BinaryDecoder.cs:line 259
>    at Opc.Ua.NodeStateCollection.LoadFromBinary(ISystemContext context, Stream istrm, Boolean updateTables) in C:\Users\gavagnaf\Desktop\UA\UA2\UA-.NETStandardLibrary-master\UA-.NETStandardLibrary-master\Stack\Opc.Ua.Core\Stack\State\NodeStateCollection.cs:line 270
>    at NetCoreConsoleClient.Program.LoadFromResource(Session session) in C:\Users\gavagnaf\Desktop\UA\UA2\UA-.NETStandardLibrary-master\UA-.NETStandardLibrary-master\SampleApplications\Samples\NetCoreConsoleClient\Program.cs:line 366}

I also tried to read the nodes using the LoadFromXml

```
            NodeStateCollection nodes = new NodeStateCollection();
            using (Stream stream = new FileStream("Programming.NodeSet2.xml", FileMode.Open))
            {
                try
                {
                    nodes.LoadFromXml(session.SystemContext, stream, true);
                }
                catch (Exception e)
                {
                    Console.WriteLine(e.Message);
                }
            }
```

but this time I get the following exception:

> {Opc.Ua.ServiceResultException: Could not resolve namespace uri: http://opcfoundation.org/UA/2011/03/UANodeSet.xsd
>    at Opc.Ua.NodeState.LoadNode(ISystemContext context, XmlDecoder decoder) in C:\Users\gavagnaf\Desktop\UA\UA2\UA-.NETStandardLibrary-master\UA-.NETStandardLibrary-master\Stack\Opc.Ua.Core\Stack\State\NodeState.cs:line 1604
>    at Opc.Ua.NodeStateCollection.LoadFromXml(ISystemContext context, Stream istrm, Boolean updateTables) in C:\Users\gavagnaf\Desktop\UA\UA2\UA-.NETStandardLibrary-master\UA-.NETStandardLibrary-master\Stack\Opc.Ua.Core\Stack\State\NodeStateCollection.cs:line 384
>    at NetCoreConsoleClient.Program.LoadFromXml(Session session) in C:\Users\gavagnaf\Desktop\UA\UA2\UA-.NETStandardLibrary-master\UA-.NETStandardLibrary-master\SampleApplications\Samples\NetCoreConsoleClient\Program.cs:line 383} 

Instead, these method work properly:

```
            NodeStateCollection nodes = new NodeStateCollection();
            using (Stream stream = new FileStream("Programming.uanodes", FileMode.Open))
            {
                try
                {
                    nodes.LoadFromBinary(session.SystemContext, stream, true);
                }
                catch (Exception e)
                {
                    Console.WriteLine(e.Message);
                }
            }
            NodeStateCollection xmlNodes = new NodeStateCollection();
            using (Stream stream = new FileStream("Programming.xml", FileMode.Open))
            {
                try
                {
                    xmlNodes.LoadFromXml(session.SystemContext, stream, true);
                }
                catch (Exception e)
                {
                    Console.WriteLine(e.Message);
                }
            }
```

I'm probably doing something wrong in the first two cases, so could you provide a quick example of save/load a NodeSet2.xml file? 

Thanks in advance

AlinMoldovean, 2017-11-22:
Hi @CodingHarry ,

Maybe you can try the following code for loading a UANodeSet from a NodeSet2 stream:
Opc.Ua.Export.UANodeSet nodeSet = Opc.Ua.Export.UANodeSet.Read(istrm);

CodingHarry, 2017-11-23:
@AlinMoldovean just to let you know

On going. Currently our server does not support an export to get the NodeSet2.xml. I already made a support request at the manufacturer and hope this feature will be implemented soon.

closed 2018-02-07 by AlinMoldovean