OPCFoundation/UA-.NETStandard #294: Alarms and Conditions sample

christophsh, 2017-12-06:
Can anyone point me to a sample/or show me how to implement Alarms and Conditions in a minimalist way? The only requirement would be to retrieve any kind of event data by creating a 'subscription' to it before. I already created an opc-client class by following the given console sample. However i can't figure out how to retrieve events with this specific opc implementation.

Thanks for any help on this :+1:

cemremengu, 2017-12-18:
+1 for this, a simple example for monitoring alarms would be great

awcullen, 2017-12-19:
There are three steps to monitor events.
First, define an array named "selectClause" with the event fields you wish to receive. You can choose any fields from any event type. If the server raises  an event that does not have the field you request, you just receive a null value for the requested field.

Second, add a monitored item to a subscription like you do with variables... mostly.  The node has to be an object, the atttributeId is EventNotifier, the sampling interval is -1, and EventFilter set with the selectClause .

Lastly, when the item Notification event is raised, be prepared to dequeue a list of EventFields. There will be a Variant for each field of the selectClause. Some Variant values may be null.

```
            Console.WriteLine("5b - Define the event fields you want to receive.");
            var selectClause = new[] {
                new SimpleAttributeOperand { TypeDefinitionId = ObjectTypeIds.BaseEventType, BrowsePath = new[] { new QualifiedName("EventId") }, AttributeId = Attributes.Value },
                new SimpleAttributeOperand { TypeDefinitionId = ObjectTypeIds.BaseEventType, BrowsePath = new[] { new QualifiedName("EventType") }, AttributeId = Attributes.Value },
                new SimpleAttributeOperand { TypeDefinitionId = ObjectTypeIds.BaseEventType, BrowsePath = new[] { new QualifiedName("SourceName") }, AttributeId = Attributes.Value },
                new SimpleAttributeOperand { TypeDefinitionId = ObjectTypeIds.BaseEventType, BrowsePath = new[] { new QualifiedName("Time") }, AttributeId = Attributes.Value },
                new SimpleAttributeOperand { TypeDefinitionId = ObjectTypeIds.BaseEventType, BrowsePath = new[] { new QualifiedName("Message") }, AttributeId = Attributes.Value },
                new SimpleAttributeOperand { TypeDefinitionId = ObjectTypeIds.BaseEventType, BrowsePath = new[] { new QualifiedName("Severity") }, AttributeId = Attributes.Value }
            };

            Console.WriteLine("5 - Create a subscription with publishing interval of 1 second.");
            var subscription = new Subscription(session.DefaultSubscription) { PublishingInterval = 1000 };

            Console.WriteLine("6 - Add a list of items to the subscription.");
            var list = new List<MonitoredItem> {
                new MonitoredItem(subscription.DefaultItem)
                {
                    NodeClass = NodeClass.Object,
                    StartNodeId = ObjectIds.Server,
                    AttributeId = Attributes.EventNotifier,
                    SamplingInterval = -1,
                    QueueSize = 100,
                    CacheQueueSize = 100,
                    Filter=new EventFilter() { SelectClauses = selectClause}
                }
            };
            list.ForEach(i => i.Notification += OnNotification);
            subscription.AddItems(list);

            Console.WriteLine("7 - Add the subscription to the session.");
            session.AddSubscription(subscription);
            subscription.Create();

        private static void OnNotification(MonitoredItem item, MonitoredItemNotificationEventArgs e)
        {
            foreach (var value in item.DequeueEvents())
            {
                Console.WriteLine("{3}: {2}, {4}, {5}", value.EventFields.Select(v => v.Value ?? "<null>").ToArray());
            }
        }
```

closed 2017-12-19 by christophsh

nour1876, 2023-03-27:
Hi i actually want to add ondatachange alarm on a tag in the server , i actually have the addressSpace displayed on server how can i do that can you please help me i'm actually working on opc foundation .NetStandard (opc ua)