OPCFoundation/UA-.NETStandard #295: GDS question

eoursel, 2017-12-06:
Did you tested the following scenario for the GDS?
- StartNewKeyPairRequest
- FinishKeyPairRequest

At the end i have a DER file and a PFX file. It's look ok but if you view the certificate with the windows tools it says that the digital signature is invalid. Also an OPCUA server do not validate the certificate with a digital signature invalid error. 

I am working on the branch gds_work_certs  from @mregen.

mregen, 2017-12-07:
@eoursel Thanks for the feedback, I will double check the use case. Did you use the gds client app or a different tool?

eoursel, 2017-12-08:
Hi @mregen, I wrote a .NETCORE console App with a command line Library. The source code is inspired from the GdsClientTest. It is a quick hack for testing. I don’t like so much the GDSClient Winform applcation. I would prefer a ASP.NET CORE application which could be deployed in docker containers or as an Azure application. Perhaps that is Something we could add to the roadmap, 

I am fighting with certificates. How do you generate your certificates for the NETCORE server ?  I have a binary version of Opc.Ua.CertificateGenerator which generate SHA1 signed certificates which are rejected  by the Opc UA stack.  I expect to use certificates signed by a chain of CA.  

Regards, 

Eric

Provenance : Courrier pour Windows 10

De : Martin Regen
Envoyé le :jeudi 7 décembre 2017 10:47
À : OPCFoundation/UA-.NETStandard
Cc : Eric OURSEL; Mention
Objet :Re: [OPCFoundation/UA-.NETStandard] GDS question (#295)

@eoursel Thanks for the feedback, I will double check the use case. Did you use the gds client app or a different tool?

mregen, 2017-12-08:
Hi @eoursel, there are a few samples which implement a .Net Core server in the repo, e.g. CoreReferenceServer. Each of them automatically create the self signed cert with recommend key length (2048) and SHA256 (no need for without the Certificate generator). There are flags in the securityconfiguration to reenable shorter key length or SHA1 for a server.
With the GDS you can create *only* certs signed by a CA. You can take a look at the unit tests in the GDS project how to create key pairs etc. Its fairly straighforward.

Hope this helps..

Martin

eoursel, 2017-12-08:
Hi again @mregen, 

Yes I am aware about the certifcates generated by the .NET Core server. Because i would like to test certifcates issued by the GDS, I tried to use the GDS to issue these certificates. That’s exactly what i have done, i took your unit tests to see how to get  key pairs. I do have DER and PFX files but the server says the digital signature of the cert is bad. 

I noticed that in your unit test, you don’t check that the returned certificate is valid or not. You check only that you have a certificate which is returned. 

  [Test, Order(510)]
        public void FinishGoodNewKeyPairRequests()
        {
            AssertIgnoreTestWithoutGoodRegistration();
            ConnectGDS(true);
            bool requestBusy;
            do
            {
                requestBusy = false;
                foreach (var application in _goodApplicationTestSet)
                {
                    if (application.CertificateRequestId != null)
                    {
                        byte[] certificate = _gdsClient.GDSClient.FinishRequest(
                            application.ApplicationRecord.ApplicationId,
                            application.CertificateRequestId,
                            out byte[] privateKey,
                            out byte[][] issuerCertificates
                            );

                        if (certificate != null)
                        {
                            Assert.NotNull(certificate);
                            Assert.NotNull(privateKey);
                            Assert.NotNull(issuerCertificates);
                            application.Certificate = certificate;
                            application.PrivateKey = privateKey;
                            application.IssuerCertificates = issuerCertificates;
                            application.CertificateRequestId = null;
                            // TODO: verify cert subject and extensions
                        }
                        else
                        {
                            requestBusy = true;
                        }
                    }
                }

                if (requestBusy)
                {
                    Thread.Sleep(500);
                }
            } while (requestBusy);
        }
Provenance : Courrier pour Windows 10

De : Martin Regen
Envoyé le :vendredi 8 décembre 2017 14:20
À : OPCFoundation/UA-.NETStandard
Cc : Eric OURSEL; Mention
Objet :Re: [OPCFoundation/UA-.NETStandard] GDS question (#295)

Hi @eoursel, there are a few samples which implement a .Net Core server in the repo, e.g. CoreReferenceServer. Each of them automatically create the self signed cert with recommend key length (2048) and SHA256 (no need for without the Certificate generator). There are flags in the securityconfiguration to reenable shorter key length or SHA1 for a server.
With the GDS you can create only certs signed by a CA. You can take a look at the unit tests in the GDS project how to create key pairs etc. Its fairly straighforward.
Hope this helps..
Martin

mregen, 2017-12-08:
Hi @eoursel, I added a cert verification to the unit test. Looks all good. Maybe you try the GDS client next and compare the output with your client tool..

eoursel, 2017-12-10:
Hi @mregen 

I just modified your unit test to write the certificate on the filesystem to display it with Windows. As you can in the screenshot, there is an error detected by Windows about the validity of the digital signature. Before doing that, I installed the GDS Test CA in the trusted root certificates. 

  ```
byte[] exportedCert = newCert.Export(X509ContentType.Cert);
  string file = String.Format("cert-{0}.der", newCert.Subject);
  File.WriteAllBytes(file, exportedCert);
```
[image]
[image]

Regards
Eric

mregen, 2017-12-11:
Hi @eoursel , I try to repro here and I think I found the disconnect.
The test runner creates a new CA certificate on every pass. So all looks good after the first run I imported the CA cert to the trusted root store:
[image]

then I run one more time and look at the new certs (created by the test runner with a new CA authority with the same name)
[image]
--> Now the cert is 'invalid' because I didn't update the root CA. Makes sense?

closed 2018-01-02 by mregen