OPCFoundation/UA-.NETStandard #302: Question? - Limit Nodes Shown to Connecting Clients - UWP IOT Windows Core

Cur001, 2017-12-18:
Morning guys, 
I am wondering if anyone can point me towards how i would limit the nodes that are visible to connecting clients based on authentication provided. i have the authentication stuff done so can tell who is trying to connect to the server but i cant find the code needed to limit the nodes that are visible.

Any help appreciated

AlinMoldovean, 2017-12-19:
Hello @Cur001,

In order to implement a user authorization you can implement an override of the specific service calls you are intending to filter.
This can be done in the class extending StandardServer, e.g. ReferenceServer
The identity of the user behind the connected client can be evaluated from the context of the service call.

One example for limiting the results of Browse calls:

[image]

public override ResponseHeader Browse(
            RequestHeader requestHeader,
            ViewDescription view,
            uint requestedMaxReferencesPerNode,
            BrowseDescriptionCollection nodesToBrowse,
            out BrowseResultCollection results,
            out DiagnosticInfoCollection diagnosticInfos)
        {
            results = null;
            diagnosticInfos = null;

            OperationContext context = ValidateRequest(requestHeader, RequestType.Browse);

            try
            {

                if (nodesToBrowse == null || nodesToBrowse.Count == 0)
                {
                    throw new ServiceResultException(StatusCodes.BadNothingToDo);
                }

                // return empty browse results for Anonymous users
                // This logic should be further extended....
                if (context != null & context.UserIdentity.TokenType == UserTokenType.Anonymous)
                {
                    results = new BrowseResultCollection(nodesToBrowse.Count);
                    diagnosticInfos = new DiagnosticInfoCollection(nodesToBrowse.Count);
                }
                else
                {
                    ServerInternal.NodeManager.Browse(
                    context,
                    view,
                    requestedMaxReferencesPerNode,
                    nodesToBrowse,
                    out results,
                    out diagnosticInfos);
                }

                return CreateResponse(requestHeader, context.StringTable);
            }
            catch (ServiceResultException e)
            {
                lock (ServerInternal.DiagnosticsWriteLock)
                {
                    ServerInternal.ServerDiagnostics.RejectedRequestsCount++;

                    if (IsSecurityError(e.StatusCode))
                    {
                        ServerInternal.ServerDiagnostics.SecurityRejectedRequestsCount++;
                    }
                }

                throw TranslateException(context, e);
            }
            finally
            {
                OnRequestComplete(context);
            }
        }

Cur001, 2017-12-19:
Thanks for the input @AlinMoldovean I will have a look at adding same to my solution 👍

AlinMoldovean, 2017-12-19:
Hi @Cur001,

Yes, you can also filter results and diagnosticInfos after ServerInternal.NodeManager.Browse...
Make sure they will have the same length

Cur001, 2017-12-19:
@AlinMoldovean - thanks

AlinMoldovean, 2017-12-20:
@Cur001 - what is the actual question?

Cur001, 2017-12-20:
Actually i figured it out thanks for your help

closed 2017-12-20 by AlinMoldovean