OPCFoundation/UA-.NETStandard #303: Question - NodeManager.Browse Identifying Node

Cur001, 2017-12-20:
Hi Guys,
I am working with the .Net SDK for both windows and UWP I am looking to limit the visibility of nodes based on user authentication. I have my Server Class which inherits from StandardServer and a browse override in this class.

I am creating variables which sit inside folders via:
` private BaseDataVariableState CreateVariable(NodeState parent, string path, string name, NodeId dataType, int valueRank)
        {
            BaseDataVariableState variable = new BaseDataVariableState(parent);

            variable.SymbolicName = name;
            variable.ReferenceTypeId = ReferenceTypes.Organizes;
            variable.TypeDefinitionId = VariableTypeIds.BaseDataVariableType;
            variable.NodeId = new NodeId(path, NamespaceIndex);
            variable.BrowseName = new QualifiedName(path, NamespaceIndex);
            variable.DisplayName = new LocalizedText("en", name);
            variable.WriteMask = AttributeWriteMask.DisplayName | AttributeWriteMask.Description;
            variable.UserWriteMask = AttributeWriteMask.DisplayName | AttributeWriteMask.Description;
            variable.DataType = dataType;
            variable.ValueRank = valueRank;
            variable.AccessLevel = AccessLevels.CurrentReadOrWrite;
            variable.UserAccessLevel = AccessLevels.CurrentReadOrWrite;
            variable.Historizing = false;
            variable.Value = m_system.ReadValue(variable);
            variable.StatusCode = StatusCodes.Good;
            variable.Timestamp = DateTime.UtcNow;
            variable.OnReadValue = DoSystemRead;
            

            if (parent != null)
            {
                parent.AddChild(variable);
            }

            return variable;
        }`

my variables have an object inside the .Handle property that has a Guid Identifier a AccessLevel enum and a Index if multiple versions of the variable exist.

I am being told that Browse in the class that inherits from StandardServer is where i can change the resultscollection content to drop out any nodes where the client authentication is not the same as the super user in the config but i am having trouble getting back to the handle property on the node. 

Can anyone show me what i need to do with the results from this call 

    ServerInternal.NodeManager.Browse(
                    context,
                    view,
                    requestedMaxReferencesPerNode,
                    nodesToBrowse,
                    out results,
                    out diagnosticInfos);

to get the .Handle Property i have assigned to the variable when i created it.

AlinMoldovean, 2017-12-20:
Hi @Cur001,

I made a small code sample to get the attached handle for your case:

ServerInternal.NodeManager.Browse(
                    context,
                    view,
                    requestedMaxReferencesPerNode,
                    nodesToBrowse,
                    out results,
                    out diagnosticInfos);

                foreach (BrowseResult browseResult in results)
                {
                    foreach (ReferenceDescription referenceDescription in browseResult.References)
                    {
                        // search the node in address space
                        NodeId nodeId = ExpandedNodeId.ToNodeId(referenceDescription.NodeId, ServerInternal.NamespaceUris);
                        INodeManager nodeManager = null;
                        object handle = ServerInternal.NodeManager.GetManagerHandle(nodeId, out nodeManager);
                        NodeHandle nodeHandle = handle as NodeHandle;

                        if (nodeHandle != null)
                        {
                            BaseDataVariableState variable = nodeHandle.Node as BaseDataVariableState;

                            if (variable != null)
                            {
                                // Get the attached handle
                                object handleProperty = variable.Handle;
                            }
                        }
                    }
                }

Cur001, 2017-12-20:
@AlinMoldovean thanks for that, have implemented the above but my variable nodes are are constantly null when typed to BaseDataVariableState 
BaseDataVariableState variable = nodeHandle.Node as BaseDataVariableState;
as variable is always null but the nodeHandle.Node is type {[VariableType]BaseDataVariableType}
any ideas?

AlinMoldovean, 2017-12-20:
In that case just adapt the code
    NodeState node= nodeHandle.Node as BaseDataVariableState;

                            if (node != null)
                            {
                                // Get the attached handle
                                object handleProperty = node.Handle;
                            }
it depends what the GetManagerHandle() method returns on your NodeManager implementation...

Cur001, 2017-12-20:
my variable node .Handle on server startup is set to TestObjectReference object ie: 
`         var tempGuid = Guid.Parse("C60A5431-229F-456C-8142-E5EDE834C263");
                    BaseDataVariableState variable1 = CreateVariable(root, "ConnectionState", "ConnectionState", DataTypeIds.Boolean, ValueRanks.Scalar);
                    variable1.Handle = new TestObjectReference { Id = tempGuid, Index = 0, AccessLevel = UserAccessLevel.SuperUser };
                    variables.Add(variable1);`

Both versions of the code above result in a null .Handle 

The GetmanagerHandle() impletmation is as it was supplied 

`  protected virtual object GetManagerHandle(ISystemContext context, NodeId nodeId, IDictionary<NodeId,NodeState> cache)
        {
            lock (Lock)
            {
                // quickly exclude nodes that not in the namespace.
                if (!IsNodeIdInNamespace(nodeId))
                {
                    return null;
                }

                // lookup the node.
                NodeState node = null;

                if (!m_predefinedNodes.TryGetValue(nodeId, out node))
                {
                    return null;
                }

                return node;
            }
        }`

` public bool TryGetValue(NodeId key, out T value)
        {
            value = default(T);

            if (key == null)
            {
                return false;
            }

            switch (key.IdType)
            {
                case IdType.Numeric:
                {
                    ulong id = ((ulong)key.NamespaceIndex)<<32;
                    id += (uint)key.Identifier;
                    return m_numericIds.TryGetValue(id, out value);
                }

                case IdType.String:
                {
                    IDictionary<string,T> dictionary = GetStringDictionary(key.NamespaceIndex, false);

                    if (dictionary != null)
                    {
                        return dictionary.TryGetValue((string)key.Identifier, out value);
                    }

                    break;
                }

                case IdType.Guid:
                {
                    IDictionary<Guid,T> dictionary = GetGuidDictionary(key.NamespaceIndex, false);

                    if (dictionary != null)
                    {
                        return dictionary.TryGetValue((Guid)key.Identifier, out value);
                    }

                    break;
                }

                case IdType.Opaque:
                {
                    IDictionary<ByteKey,T> dictionary = GetOpaqueDictionary(key.NamespaceIndex, false);

                    if (dictionary != null)
                    {
                        return dictionary.TryGetValue(new ByteKey((byte[])key.Identifier), out value);
                    }

                    break;
                }
            }
     
            return false;
        }`

AlinMoldovean, 2017-12-20:
As I said, "it depends what the GetManagerHandle() method returns on your NodeManager implementation"
I tested against ReferenceServer and it has the following (maybe you can try it...):
 
       /// <summary>
        /// Returns a unique handle for the node.
        /// </summary>
        protected override NodeHandle GetManagerHandle(ServerSystemContext context, NodeId nodeId, IDictionary<NodeId, NodeState> cache)
        {
            lock (Lock)
            {
                // quickly exclude nodes that are not in the namespace. 
                if (!IsNodeIdInNamespace(nodeId))
                {
                    return null;
                }

                NodeState node = null;

                if (!PredefinedNodes.TryGetValue(nodeId, out node))
                {
                    return null;
                }

                NodeHandle handle = new NodeHandle();

                handle.NodeId = nodeId;
                handle.Node = node;
                handle.Validated = true;

                return handle;
            }
        }

Cur001, 2017-12-20:
@AlinMoldovean - sorry abou this maybe i am having a slow day but the call 
  object handle = ServerInternal.NodeManager.GetManagerHandle(nodeId, out nodeManager);
is routing to MasterNodeManager.cs and not to MyNodeManager which inherits from CustomNodeManager2 which now contains the above GetManagerHandle code you have supplied?

AlinMoldovean, 2017-12-20:
MasterNodeManager.GetManagerHandle() will call INodeManager.GetManagerHandle(nodeId) for registered node managers until the first != null result is returned...

CustomNodeManager2.GetManagerHandle(nodeId) will call the override for GetManagerHandle(ISystemContext context, NodeId nodeId, IDictionary<NodeId,NodeState> cache) on your MyNodeManager 

    public virtual object GetManagerHandle(NodeId nodeId)
        {
            lock (Lock)
            {
                return GetManagerHandle(m_systemContext, nodeId, null);
            }
        }

Cur001, 2017-12-20:
@AlinMoldovean - thanks for that can i ask how would you physically remove a browseresult from the results collection without introducing errors such as BadstructureMissing.

My structure is Root which all users can see inside root is variable nodes and some folder nodes with folder nodes containing a mixture of variable nodes but a folder node can contain variable nodes that may need to be hidden to some users.

I only end up with a stand alone variable or variable that a child of a folder node but does not have any children itself  BaseDataVariableState variable = nodeHandle.Node as BaseDataVariableState produces a null variable. 

at the moment from within the loop i cant seem to remove a single node but leave others without introducing BadstructureMissing errors.

AlinMoldovean, 2018-01-05:
Hi @Cur001 ,

As I also suggested for issue #308 , maybe you should try to create a new set of results instead of trying to remove elements from the complete set of results.

Cur001, 2018-01-08:
#308 - i have included the solution should it be of any use to anyone else. @AlinMoldovean thanks for your help

closed 2018-01-08 by Cur001