OPCFoundation/UA-.NETStandard #305: Certificate doesn't meet minimum key length requirement

cxd110, 2017-12-31:
I connect the OPC UA SERVER of GE CPE400, and the error is: Certificate doesn't meet minimum key length requirement, where is the problem?

mregen, 2018-01-02:
@cxd110 Sha1 signed certs and 1k key length are by default deprecated. 
Add the following lines to your `<SecurityConfiguration>` to bypass the security check:
```
    <!-- WARNING: SHA1 signed certficates are by default rejected and should be phased out.-->
    <RejectSHA1SignedCertificates>false</RejectSHA1SignedCertificates>
    <MinimumCertificateKeySize>1024</MinimumCertificateKeySize>

```

cxd110, 2018-01-02:
According to your hint, I solved my problem
thanks！

closed 2018-01-02 by cxd110

torikachi24, 2020-05-28:
@mregen Hi sir, I have same problem with 2 libraries Opc.Ua.Core and Opc.Ua.Client. I have added 2 libraries to my project with asp web form. But I don't know which file need the code to bypass the security check like you said.

AlinMoldovean, 2020-06-03:
@torikachi24 ,

You should set the `RejectSHA1SignedCertificates `flag to false in `ApplicationConfiguration` if you do not load it from an XML configuration file:

```
// load the application configuration.
application.LoadApplicationConfiguration(false).Wait();
application.ApplicationConfiguration.SecurityConfiguration.RejectSHA1SignedCertificates = false;
```