OPCFoundation/UA-.NETStandard #306: SHA1 signed certificates are not trusted

vrushalikatkade, 2018-01-02:
I have used your dotnetcore application for connecting the OPCUA server .after connecting the simense wincc server it shows the following exception"  SHA1 signed certificates are not trusted " and i also connect your OPCUA server " opc.tcp://myserver:51210/UA/SampleServer" it thows the following server "Error establishing a connection: BadNotConnected " 

[image]

i attached the exception image to the post . please suggest any solution to the post

Thanks

AlinMoldovean, 2018-01-03:
Hi @vrushalikatkade ,

Please take a look at issue #305 
There is a configuration flag for accepting SHA1 certificates:

<RejectSHA1SignedCertificates>true</RejectSHA1SignedCertificates>

vrushalikatkade, 2018-01-03:
Thanks for reply , in which file to update the <SecurityConfiguration>

AlinMoldovean, 2018-01-03:
If you are using NetCoreConsoleClient, the configuration parameters are set programmatically in program.cs
You can set SecurityConfiguration fields like this:
SecurityConfiguration = new SecurityConfiguration
                {
                    ApplicationCertificate = new CertificateIdentifier
                    {
                        StoreType = "X509Store",
                        StorePath = "CurrentUser\\My",
                        SubjectName = "UA Core Sample Client"
                    },
                    TrustedPeerCertificates = new CertificateTrustList
                    {
                        StoreType = "Directory",
                        StorePath = "OPC Foundation/CertificateStores/UA Applications",
                    },
                    TrustedIssuerCertificates = new CertificateTrustList
                    {
                        StoreType = "Directory",
                        StorePath = "OPC Foundation/CertificateStores/UA Certificate Authorities",
                    },
                    RejectedCertificateStore = new CertificateTrustList
                    {
                        StoreType = "Directory",
                        StorePath = "OPC Foundation/CertificateStores/RejectedCertificates",
                    },
                    NonceLength = 32,
                    AutoAcceptUntrustedCertificates = autoAccept,
                    RejectSHA1SignedCertificates = false,
                    MinimumCertificateKeySize = 1024
                },

vrushalikatkade, 2018-01-03:
Thanks for reply , it's working.I have second issue .after connecting server it display the this output like this
C:\Users\vrushali\Downloads\UA-.NETStandard-master\UA-.NETStandard-master\Sample
Applications\Samples\NetCoreConsoleClient>dotnet run opc.tcp://vrushali-PC:4862
.Net Core OPC UA Console Client sample
1 - Create an Application Configuration.
2 - Discover endpoints of opc.tcp://vrushali-PC:4862.
    Selected endpoint uses: Basic256
3 - Create a session with OPC UA server.
Accepted Certificate: CN=Siemens OPC UA Server for WinCC, OU=I IA AS, O=Siemens
AG, S=BW, C=DE, DC=vrushali-PC
4 - Browse the OPC UA server namespace.
 DisplayName, BrowseName, NodeClass
 Server, Server, Object
   + ServerArray, ServerArray, Variable
   + NamespaceArray, NamespaceArray, Variable
   + ServerStatus, ServerStatus, Variable
   + ServiceLevel, ServiceLevel, Variable
   + Auditing, Auditing, Variable
   + ServerCapabilities, ServerCapabilities, Object
   + ServerDiagnostics, ServerDiagnostics, Object
   + VendorServerInfo, VendorServerInfo, Object
   + ServerRedundancy, ServerRedundancy, Object
   + GetMonitoredItems, GetMonitoredItems, Method
 WinCC, 1:WinCC, Object
   + @LOCALMACHINE::, 1:@LOCALMACHINE::, Object
5 - Create a subscription with publishing interval of 1 second.
6 - Add a list of items (server current time and status) to the subscription.
7 - Add the subscription to the session.
8 - Running...Press Ctrl-C to exit...
ServerStatusCurrentTime: 2018-01-03 09:34:00, 2018-01-03 09:34:00, Good
ServerStatusCurrentTime: 2018-01-03 09:34:01, 2018-01-03 09:34:01, Good
ServerStatusCurrentTime: 2018-01-03 09:34:02, 2018-01-03 09:34:02, Good
ServerStatusCurrentTime: 2018-01-03 09:34:03, 2018-01-03 09:34:03, Good
ServerStatusCurrentTime: 2018-01-03 09:34:04, 2018-01-03 09:34:04, Good
ServerStatusCurrentTime: 2018-01-03 09:34:05, 2018-01-03 09:34:05, Good
ServerStatusCurrentTime: 2018-01-03 09:34:06, 2018-01-03 09:34:06, Good
ServerStatusCurrentTime: 2018-01-03 09:34:07, 2018-01-03 09:34:07, Good
ServerStatusCurrentTime: 2018-01-03 09:34:08, 2018-01-03 09:34:08, Good

how to browse the server & create the subscription ? OPCUA DA,HDA ,AE  is developed in this dotnet core application ? how to use it? 

I tried to add the option 1,2,3, from output list but it's not working.

mregen, 2018-02-06:
@vrushalikatkade the console sample has been restructured to make it easier to configure. sha1 should be solved now. Please open a new issue for a new topic...

closed 2018-02-06 by mregen

andreaballerini, 2018-03-15:
Hi @AlinMoldovean 
I have the same problem of @vrushalikatkade 
i do your advice and i tryed everithing but the program continue to show "SHA1 signed certificates are not trusted".

Can you help me?

AlinMoldovean, 2018-03-15:
Hi @andreaballerini ,

The server you are connecting is built with this UA NET Standard Stack?
If yes, please set the "RejectSHA1SignedCertificates" configuration parameter to false: 

```
  <SecurityConfiguration>
    
    <!-- Where the application instance certificate is stored (MachineDefault) -->
    <ApplicationCertificate>
      <StoreType>X509Store</StoreType>
      <StorePath>CurrentUser\UA_MachineDefault</StorePath>
      <SubjectName>CN=UA Sample Server, C=US, S=Arizona, O=OPC Foundation, DC=localhost</SubjectName>
    </ApplicationCertificate>

    <!-- Where the issuer certificate are stored (certificate authorities) -->
    <TrustedIssuerCertificates>
      <StoreType>Directory</StoreType>
      <StorePath>%CommonApplicationData%\OPC Foundation\CertificateStores\UA Certificate Authorities</StorePath>
    </TrustedIssuerCertificates>
    
    <!-- Where the trust list is stored (UA Applications) -->
    <TrustedPeerCertificates>
      <StoreType>Directory</StoreType>
      <StorePath>%CommonApplicationData%\OPC Foundation\CertificateStores\UA Applications</StorePath>
    </TrustedPeerCertificates>
    
    <!-- The directory used to store invalid certficates for later review by the administrator. -->
    <RejectedCertificateStore>
      <StoreType>Directory</StoreType>
      <StorePath>%CommonApplicationData%\OPC Foundation\CertificateStores\RejectedCertificates</StorePath>
    </RejectedCertificateStore>

    <RejectSHA1SignedCertificates>false</RejectSHA1SignedCertificates>

  </SecurityConfiguration>
```

andreaballerini, 2018-03-15:
Hi @AlinMoldovean , thanks for your answer.
i do that and now i have this problem:

_"Error establishing a connection: Error received from remote host: Could not verify security on OpenSecureChannel request."_

There is some doc where i can read the correct settings?
Because on other server opc i need to insert credentials and i don't know how do it.

Thanks you

AlinMoldovean, 2018-03-16:
Hi @andreaballerini ,

It seems that the certificate of the client is not trusted by the server.
You need to copy the client certificate of the client in the trust list of the server and the certificate of the server in the trust list of the server.

andreaballerini, 2018-03-19:
Ok thanks it works!