OPCFoundation/UA-.NETStandard #307: Mimimal OPC UA Console Client using UA.-NETStandard

MrPoudel, 2018-01-03:
Hi everybody,
now I'm using OPC Foundation's  UA-.NETStandard library to test the availability of the OPC UA server over the network. If I disconnect the device( in my case raspberrypi hosting a OPC UA server) from the network it should immediately detect the disconnection. When I connect the device back to the network it should detect the device immediately. Could anybody provide me a minimal example for this case.

Thanks in Advance.

MrPoudel, 2018-01-03:
@AlinMoldovean @mregen

AlinMoldovean, 2018-01-05:
Hi @MrPoudel,

OPC UA protocol does not have a predefined and built-in mechanism for a fast detection of connection lost.
The client has only a default keep-alive watchdog that does a cyclic read request, but that is not enough for your use-case.

closed 2018-01-25 by AlinMoldovean

MrPoudel, 2018-01-30:
@AlinMoldovean Thanks a lot.