OPCFoundation/UA-.NETStandard #308: ServerInternal.NodeManager.Browse

Cur001, 2018-01-04:
I am trying to limit what nodes that are visible by a client depending on their client authentication. I was pointed at overriding the Browse method in the class that inherits from StandardServer

I am performing this call after checking for the users autentication

   ServerInternal.NodeManager.Browse(
                   context,
                   view,
                   requestedMaxReferencesPerNode,
                   nodesToBrowse,
                   out results,
                   out diagnosticInfos);

I need to remove BrowseResult from the BrowseResultCollection results collection if the user is not authenticated to view the node. 

As soon as i touch remove anything from the results collection it introduces BadStructureErrors can anyone point me at the correct way to go about this.

3 types of user
Anon can see a limited list of nodes 
Username & Pass (normal user) can see full list of nodes except for specials
Username & Pass (special account superuser) can see full list of nodes including specials

Any help appricated.

AlinMoldovean, 2018-01-05:
Hi @Cur001,

I think the best way is to create a new set of browse result and populate it by iterating through the set of complete results, instead of trying to remove element from the collection.

Cur001, 2018-01-08:
@AlinMoldovean - rightly or wrongly i went the route of removing the nodes should the user not have the permission level needed, i have included the solution incase its of use to anyone else as it wasnt obvious that it was the ReferenceDescriptionCollection that i needed to add and remove elements from. 

`        /// <summary>
        /// Invokes the Browse service - Allows for the Visibility of Nodes based on users Permission Level (Authentication Level)
        /// </summary>
        /// <param name="requestHeader">The request header.</param>
        /// <param name="view">The view.</param>
        /// <param name="requestedMaxReferencesPerNode">The maximum number of references to return for each node.</param>
        /// <param name="nodesToBrowse">The list of nodes to browse.</param>
        /// <param name="results">The list of results for the passed starting nodes and filters.</param>
        /// <param name="diagnosticInfos">The diagnostic information for the results.</param>
        /// <returns>
        /// Returns a <see cref="T:Opc.Ua.ResponseHeader" /> object
        /// </returns>
        public override ResponseHeader Browse(
            RequestHeader requestHeader, 
            ViewDescription view, 
            uint requestedMaxReferencesPerNode, 
            BrowseDescriptionCollection nodesToBrowse, 
            out BrowseResultCollection results, 
            out DiagnosticInfoCollection diagnosticInfos)
        {

            OperationContext context = ValidateRequest(requestHeader, RequestType.Browse);
            ServerInternal.NodeManager.Browse(
                  context,
                  view,
                  requestedMaxReferencesPerNode,
                  nodesToBrowse,
                  out results,
                  out diagnosticInfos);

            foreach (BrowseResult browseResult in results)
            {
                ReferenceDescriptionCollection tempList = new ReferenceDescriptionCollection();

                foreach (ReferenceDescription referenceDescription in browseResult.References)
                {
                    // search the node in address space
                    NodeId nodeId = ExpandedNodeId.ToNodeId(referenceDescription.NodeId, ServerInternal.NamespaceUris);
                    INodeManager nodeManager = null;
                    object handle = ServerInternal.NodeManager.GetManagerHandle(nodeId, out nodeManager);
                    NodeHandle nodeHandle = handle as NodeHandle;

                    if (nodeHandle != null)
                    {
                        BaseDataVariableState variable = nodeHandle.Node as BaseDataVariableState;

                        if ((variable != null) && (variable.Handle !=null))
                        {
                            // Get the attached handle
                            TestObjectReference handleProperty = (TestObjectReference)variable.Handle;
                            
                            if(handleProperty != null)
                            {
                                Debug.WriteLine(handleProperty);
                                if(context != null)
                                {
                                    //We dont want to show Safeline Only Nodes if the user has provided Anon Authentication
                                    if (context.UserIdentity.TokenType == UserTokenType.Anonymous)
                                    {
                                        if (handleProperty.AccessLevel == UserAccessLevel.SuperUser)
                                        {
                                            tempList.Add(referenceDescription);
                                        }
                                    }
                                    //If they have provided Username we need to check if its Safeline special account.
                                    else if(context.UserIdentity.TokenType == UserTokenType.UserName)
                                    {
                                        if(context.UserIdentity.DisplayName == "SA")
                                        {
                                            var userNameIdentity = (UserNameIdentityToken)context.Session.IdentityToken;
                                            if(!userNameIdentity.DecryptedPassword.Equals("SecretPassword"))
                                            {
                                                if (handleProperty.AccessLevel == UserAccessLevel.SuperUser)
                                                {
                                                    tempList.Add(referenceDescription);
                                                }
                                            }
                                        }
                                        else
                                        {
                                            if (handleProperty.AccessLevel == UserAccessLevel.SuperUser)
                                            {
                                                tempList.Add(referenceDescription);
                                            }
                                        }
                                    }
                                }
                            }
                        }
                    }

                }
                if (tempList.Count > 0)
                {
                    foreach(ReferenceDescription refDesc in tempList)
                    {
                        browseResult.References.Remove(refDesc);
                    }
                    tempList.Clear();
                }
            }

            return CreateResponse(requestHeader, context.StringTable);
        }
`

closed 2018-01-08 by Cur001