OPCFoundation/UA-.NETStandard #310: ModelCompiler generated models

Cur001, 2018-01-04:
Hi Guys,

Given the example stack includes models generated with the ModelCompiler     [System.CodeDom.Compiler.GeneratedCodeAttribute("Opc.Ua.ModelCompiler", "1.0.0.0")] can anyone point me in the direction of same and any hints / tips on its use. 

I want to generate complex objects in code if possible so if anyone can point me at an example of generating the boiler object in c# rather than xml via the model generation that would be ideal.

Thanks

mregen, 2018-01-04:
@Cur001 Hi, there are two boiler in the sample, one is also implemented in code. Also the reference server is fully implemented as code --> https://github.com/OPCFoundation/UA-.NETStandard/blob/v1.03/SampleApplications/Workshop/Reference/Server/ReferenceNodeManager.cs

Cur001, 2018-01-08:
Thanks @mregen - I will have a look at generating the model via code but would also be interested in what tool was used to generate the .xml file with the opc xml nodes that was used in the origional soluition as if their is a gui interface model design should proove more intuitative.

mregen, 2018-01-18:
@Cur001 ModelCompiler is a command line tool hosted here: https://github.com/OPCFoundation/UA-ModelCompiler

closed 2018-02-06 by mregen

MikeK-10, 2019-01-04:
I think that the point of Cur001's question is missed. ModelComplier requires a .xml file, how can we obtain the .xml file if the tool used to configure the server does not generate in a format suitable for ModelCompiler. The data is available from the OPC UA server, is there a tool for exporting this to xml?

Cur001, 2019-01-04:
MikeK-10 - that was indeed my origional point, it seems that the xml may have been handballed as I could not find a modeller that gave me the same xml output and i tried unified / softings etc. in the end i gave up trying to use a modeller at all and created all my nodes in code giving me full ownership without dependency on another tool that is largely undocumented.

MikeK-10, 2019-01-04:
Thanks for reply but I want to read the xml data from OPC UA server (currently codesys). The data seems to be exist as I can see parts of it when connected to the server but cannot seem to get the whole?

Cur001, 2019-01-04:
that may mean the the xml itself is not structurally correct i have had it happen with xml generated in an early version of the modellers and then opened and resaved in a later version, it somehow corrupted the address space on me.

MikeK-10, 2019-01-04:
I can't say without being able to obtain the xml document, all the sdk samples seem to communicate fine in both directions but I need the .xml file to be able to create the custom types at client end.

AlinMoldovean, 2019-01-04:
@MikeK-10 ,

In order to be able to handle the custom complex types of the server on the client side, is not mandatory to get the complete NodeSet xml file(s) of the server.
You can have a look at #243 and #444

MikeK-10, 2019-01-07:
Hi, thanks for the reply. I have seen those threads but it isn't really helping, the only example I can find to create the types at run time is at https://github.com/OPCFoundation/UA-.NETStandard/issues/557 and as I mention in that thread there are some issues. In any case I don't think creating the types at run-time is the correct solution for my needs as while it would require the minimum configuration the rest of my client will become significantly more complex. Ideally I want to create all the types at design time based upon as simply as possible, as you mention in thread #243 it seems that the SDK is missing some features to make this happen? The only example I can find of creating the types at design time uses the model compiler which is why I asked about the xml file.