OPCFoundation/UA-.NETStandard #311: (context.UserIdentity.TokenType == UserTokenType.UserName) accessing password on Browse

Cur001, 2018-01-04:
I want to only present nodes to authenticated users - do i have a way of getting at the authenticated users password from within the Browse method in the class that inherits from StandardServer.

I can access the tokentype and display name but not the password unless i start using reflection is their something i am overlooking.

thanks

AlinMoldovean, 2018-01-05:
Hi @Cur001 ,

Maybe you can try checking the following field in the "context" reference of the service call:
context.Session.IdentityToken.DecryptedPassword

[image]

Cur001, 2018-01-08:
@AlinMoldovean - thanks for that i had incorrectly gone after the password via the context.UserIdentity - i have now changed it to your example - thanks

closed 2018-01-08 by Cur001