OPCFoundation/UA-.NETStandard #316: OPC UA Client read example ?

Grincas, 2018-01-08:
Hello,

I was looking for a code but in session i can't find write example. Can you show me ?

Grincas, 2018-01-08:
Like write value to node

AlinMoldovean, 2018-01-08:
Hi @Grincas ,

it should be something like this:

 try
            {
                WriteValue nodeToWrite = new WriteValue();
                nodeToWrite.NodeId = m_variableId;
                nodeToWrite.AttributeId = Attributes.Value;
                nodeToWrite.Value = new DataValue();
                nodeToWrite.Value.WrappedValue = GetValue();

                WriteValueCollection nodesToWrite = new WriteValueCollection();
                nodesToWrite.Add(nodeToWrite);

                // read the attributes.
                StatusCodeCollection results = null;
                DiagnosticInfoCollection diagnosticInfos = null;

                ResponseHeader responseHeader = m_session.Write(
                    null,
                    nodesToWrite,
                    out results,
                    out diagnosticInfos);

                ClientBase.ValidateResponse(results, nodesToWrite);
                ClientBase.ValidateDiagnosticInfos(diagnosticInfos, nodesToWrite);

                // check for error.
                if (StatusCode.IsBad(results[0]))
                {
                    throw ServiceResultException.Create(results[0], 0, diagnosticInfos, responseHeader.StringTable);
                }
            }
            catch (Exception exception)
            {
                ClientUtils.HandleException(this.Text, exception);
            }

closed 2018-01-25 by AlinMoldovean