OPCFoundation/UA-.NETStandard #317: Client Connect: URL not found Message - Question

Cur001, 2018-01-09:
Morning guys,

I am using UA Expert to test my OPC-UA server i am having a dialog show when i try to connect to my OPC-UA server.  There was no exact match for the specified URL in the server's endpoints, using the first server certificate found for connecting.  

I am unsure if this message would point at an issue with the ApplicationUri or the Enpoint address, Would anyone have any ideas where to breakpoint on my server for the initial connect attempt by clients? 

regards

AlinMoldovean, 2018-01-09:
Hi @Cur001 ,

The error seems to be generated on the client side. 
So the client should be debugged if possible.
It does some validation of the server certificate. 
The complete error message should provide a hint about what is wrong anyway.

Cur001, 2018-01-09:
I will raise the query with Unified Automation and see if they can highlight any issues.

closed 2018-01-09 by Cur001