OPCFoundation/UA-.NETStandard #321: Client reconnection  after loosing Server

plmonnier, 2018-01-10:
Hello, 

I'm trying to implement reconnection for my OC UA client using UA-.NetStandard and I 'm face to an exception ( ***EXCEPTION*** BadCertificateHostNameInvalid )

* Scenario to recur the problem

_Note_ : the PC must have a domains/ dns with "." like PASCAL01.SOCIETE.LOCAL

1. I run my OPC UA Server (Prosys Server in my case)
2. I run my  OPC UA client 
3. I  connect my client to Server.
4. I stop the Server . At this step my client try to reconnect unsuccesfully (normal !)
5.  I launch the sever again 
6. I have the following **EXCEPTION*** BadCertificateHostNameInvalid  when it tries to reconnect

*  The problem

The problem comes from the function **private static void CheckCertificateDomain(ConfiguredEndpoint endpoint) in Session.cs**

Let's take an example  we have the domain : PASCAL01.SOCIETE.LOCAL 
So in my Server Certificate I have DC = PASCAL01.SOCIETE.LOCAL 

When the function tries to check the domain, it compares two varaibales 
* The first one given by ` IList<string> domains = Utils.GetDomainsFromCertficate(serverCertificate);` line 829 of session.cs ) which is my case gives  **domains = {"PASCAL01.SOCIETE.LOCAL"}**

* The second one given by `string [] hostnames = endpoint.EndpointUrl.DnsSafeHost.Split('.');`  which in my case gives **hostnames = {"PASCAL01", "SOCIETE", "LOCAL"}**

So the comparison in line 844 ` if (String.Compare(hostname, domains[ii], StringComparison.CurrentCultureIgnoreCase) == 0)` in line 846 in session.cs **always fail**

This has been introduced by the commit [26881aab7a2e12d852340d7fb662f822afa60e05](https://github.com/OPCFoundation/UA-.NETStandard/commit/26881aab7a2e12d852340d7fb662f822afa60e05) in Session.cs file  

May be the .Split('.') in line 833 of Session.cs file must be removed ? 
Before this commit the split didn't exist. I don't know why it was introduce so ?

Thanks

mregen, 2018-01-24:
Hi @plmonnier, thanks for pointing out the issue. The split is indeed not quite right for your use case. 
There were some cases fixed with a split, but we need to support all properly. I look into a fix for all.

mregen, 2018-01-25:
@plmonnier Please retest your use case with latest update. Thanks!

closed 2018-01-25 by mregen

plmonnier, 2018-02-08:
@mregen I retest and it works fine for me. Thanks for the fix.