OPCFoundation/UA-.NETStandard #323: Monitor subscription data change

vrushalikatkade, 2018-01-11:
Dear Support,

        I want access the datachange event from the windows service . can you please tell me how to access the OPCUA DA data change in windows service for multiple items ? i have refer the .net core  console application   but it will be attached the single monitor item from subscription. how to apply for multiple items of subcription?
  var subscription = new Subscription(session.DefaultSubscription) { PublishingInterval = 1000 };
i refer the below code 
        
            var list = new List<MonitoredItem> {
                new MonitoredItem(subscription.DefaultItem)
                {
                    DisplayName = "ServerStatusCurrentTime", StartNodeId = "i="+Variables.Server_ServerStatus_CurrentTime.ToString()
                }
            };
            list.ForEach(i => i.Notification += OnNotification);
            subscription.AddItems(list);         
            session.AddSubscription(subscription);
            subscription.Create();

Thanks,
vrushali

AlinMoldovean, 2018-01-11:
Hi @vrushalikatkade ,

Multiple monitored items can be added in a subscription.
You only need to add more "MonitoredItem" objects in the "list" from your example above.

vrushalikatkade, 2018-01-12:
I will notify the data change of MonitoredItem one by one in events ,
is it possible to return the data change in list not one after another?

AlinMoldovean, 2018-01-12:
Hi @vrushalikatkade ,

In that case the "Subscription.FastDataChangeCallback" can be used.
It will provide the complete list of DataChange notifications from the publish response.

closed 2018-01-29 by AlinMoldovean