OPCFoundation/UA-.NETStandard #327: NullReferenceException for Enum Collection in BinaryEncoder

Noires, 2018-01-15:
Hello together,

i get an null reference exception in the BinaryEncoder of the opc ua stack, if i want to send a list of enum values from the server to the opc ua client. The issue does not appear with a array of the enum values, only if i use the generated collection of the modeler with the enum vales. 

[image]

Greetings Noires

mregen, 2018-01-15:
@Noires please provide some additional info, e.g. dimension and modeler output and type of the valueToEncode object. Internally all enumerations are represented as array, so its unclear how this can happen.
Thanks,
Martin

Noires, 2018-01-18:
It is an collection with a count of 3 items with the enum values [1,3,8].
`  [System.CodeDom.Compiler.GeneratedCodeAttribute("Opc.Ua.ModelCompiler", "1.0.0.0")]
    [DataContract(Namespace = Test.Namespace)]
    public enum TestEnumeration
    {
        [EnumMember(Value = "TEST_0")]
        TEST_0 = 0,

        [EnumMember(Value = "TEST_1")]
        TEST_1 = 1,

        [EnumMember(Value = "TEST_2")]
        TEST_2 = 2,

        [EnumMember(Value = "TEST_3")]
        TEST_3 = 3,

        [EnumMember(Value = "TEST_4")]
        TEST_4 = 4,

        [EnumMember(Value = "TEST_5")]
        TEST_5 = 5,

        [EnumMember(Value = "TEST_6")]
        TEST_6 = 6,

        [EnumMember(Value = "TEST_7")]
        TEST_7 = 7,

        [EnumMember(Value = "TEST_8")]
        TEST_8 = 8,
    }

    [System.CodeDom.Compiler.GeneratedCodeAttribute("Opc.Ua.ModelCompiler", "1.0.0.0")]
    [CollectionDataContract(Name = "ListOfTestEnumeration", Namespace = Test.Namespace, ItemName = "TestEnumeration")]
    #if !NET_STANDARD
    public partial class TestEnumerationCollection : List<TestEnumeration>, ICloneable
    #else
    public partial class TestEnumerationCollection : List<TestEnumeration>
    #endif
    {
        #region Constructors
        /// <summary>
        /// Initializes the collection with default values.
        /// </summary>
        public TestEnumerationCollection() {}

        /// <summary>
        /// Initializes the collection with an initial capacity.
        /// </summary>
        public TestEnumerationCollection(int capacity) : base(capacity) {}

        /// <summary>
        /// Initializes the collection with another collection.
        /// </summary>
        public TestEnumerationCollection(IEnumerable<TestEnumeration> collection) : base(collection) {}
        #endregion

        #region Static Operators
        /// <summary>
        /// Converts an array to a collection.
        /// </summary>
        public static implicit operator TestEnumerationCollection(TestEnumeration[] values)
        {
            if (values != null)
            {
                return new TestEnumerationCollection(values);
            }

            return new TestEnumerationCollection();
        }

        /// <summary>
        /// Converts a collection to an array.
        /// </summary>
        public static explicit operator TestEnumeration[](TestEnumerationCollection values)
        {
            if (values != null)
            {
                return values.ToArray();
            }

            return null;
        }
        #endregion

        #if !NET_STANDARD
        #region ICloneable Methods
        /// <summary>
        /// Creates a deep copy of the collection.
        /// </summary>
        public object Clone()
        {
            return (TestEnumerationCollection)this.MemberwiseClone();
        }
        #endregion
        #endif
`

AlinMoldovean, 2018-05-17:
could not reproduce the problem.

closed 2018-05-17 by AlinMoldovean