OPCFoundation/UA-.NETStandard #328: Support for StartAsService or an alternative method

gavioto, 2018-01-16:
How can I use the old Sample Projects as a Windows Service?

I have an old UA application using net46 and Native x86 dll that I would like to migrate to NetStandard and execute as a Windows Service (doing the less possible effort..). It's derived from the old Reference Server, but now I see that 350.6 doesn't support ProccessArguments() and StartAsService().

I've seen that it exist in ApplicationInstance on "master", but it is not implemented yet.

AlinMoldovean, 2018-01-23:
Hi @gavioto 

This .NETStandard UA stack does not have integrated support for running as Windows Service because is for multi-platform.

You will need to configure a windows service application project and run the client or server component inside it.
Something like this: 

       class DAClientService : ServiceBase
    {
        #region Constructors

        /// <summary>
        /// Initializes a new instance of the DAClientService class.
        /// </summary>
        public DAClientService(ApplicationConfiguration configuration)
        {
            m_configuration = configuration;
            base.ServiceName = "DataAccess Client Service";
        }

        #endregion

        #region Overridden Methods

        /// <summary>
        /// Executed when a Start command is sent to the service
        /// </summary>
        /// <param name="args"></param>
        protected override void OnStart(string[] args)
        {
            base.OnStart(args);

            // Init DAClient
            m_DAClient = new DAClient(m_configuration);
        }

        /// <summary>
        /// Executed when a Stop command is sent to the service
        /// </summary>
        protected override void OnStop()
        {
            base.OnStop();
            
            // Dispose DA Client
            if (m_DAClient != null)
            {
                if (m_DAClient.Session != null )
                {
                    m_DAClient.Session.Dispose();
                }

                m_DAClient = null;
            }
        }
...

closed 2018-02-07 by AlinMoldovean

gavioto, 2018-02-20:
Thankyou for your response @AlinMoldovean .