OPCFoundation/UA-.NETStandard #331: Subscription FastDataChangeCallback in windows service

vrushalikatkade, 2018-01-24:
Dear support,

      I have used the following code in windows service .but Subscription FastDataChangeCallback  not invoke in windows service .

/// Session connection
    var selectedEndpoint = CoreClientUtils.SelectEndpoint(Convert.ToString(dr["endpointsurl"]), haveAppCertificate, 15000);
                if (Convert.ToString(dr["securitymode"]) == MessageSecurityMode.None.ToString())
                {
                    selectedEndpoint.SecurityMode = (MessageSecurityMode.None);
                }
                selectedEndpoint.SecurityPolicyUri = Convert.ToString(dr["securitipolicies"]);
              

                var endpointConfiguration = EndpointConfiguration.Create(config);
                var endpoint = new ConfiguredEndpoint(null, selectedEndpoint, endpointConfiguration);
                var session = await Session.Create(config, endpoint, false, "OPCUA service", 70000, new UserIdentity(new AnonymousIdentityToken()), null);

              

  if (session .Connected)
                {
                    Subscription m_subscription = new Subscription();
                    m_subscription.Handle = Convert.ToInt32(dr["subID"]);
                    m_subscription.DisplayName = Convert.ToString(dr["SubName"]);
                    m_subscription.PublishingInterval = Convert.ToInt32(dr["refreshRate"]);
                    m_subscription.LifetimeCount = 1000;
                    m_subscription.KeepAliveCount = 10;
                    //  m_subscription.CurrentPublishingEnabled = true;
                    m_subscription.Priority = 255;
                   
                  

                    DataSet dsparam = param.getallparameterFsub(Convert.ToInt32(dr["subID"]));

                    List<MonitoredItem> list = new List<MonitoredItem>() ;
                    foreach (DataRow drparam in dsparam.Tables[0].Rows)
                    {

                      list  = new List<MonitoredItem> {
                new MonitoredItem()
                {
                   AttributeId=Convert.ToUInt32(drparam["atttributeID"]) ,    DisplayName = Convert.ToString(drparam["Tag_name"]),     Handle=Convert.ToInt32(drparam["itemID"]) ,MonitoringMode =MonitoringMode.Sampling
                       

                }
            }; 
                        m_subscription.AddItems(list);
                    }
                

                    m_subscription.FastDataChangeCallback +=new Opc.Ua.Client.FastDataChangeNotificationEventHandler( FastDataChangeNotificationEventHandler);
                   session .AddSubscription(m_subscription);

                    m_subscription.Create();

//FastDataChangeNotificationEventHandler
  private void FastDataChangeNotificationEventHandler(Subscription subscription, DataChangeNotification notification, IList<string> stringTable)
        {
          

            try
            {

               
                foreach (MonitoredItemNotification itemNotification in notification.MonitoredItems)
                {
                    MonitoredItem monitoredItem = subscription.FindItemByClientHandle(itemNotification.ClientHandle);

                    if (monitoredItem == null)
                    {
                        continue;
                    }

                }
              
                //  dataGridView1.Invoke(new Action(() => dataGridView1.Rows.Add(dr)));
            }
            catch (Exception exception)
            {
                //ClientUtils.HandleException(this.Text, exception);
            }
        }

Thanks

vrushalikatkade, 2018-01-25:
plz reply

AlinMoldovean, 2018-01-25:
Hi @vrushalikatkade,

The problem should not be related to application running as windows service.
The problem seems to be that the publishing is not enable for the created subscription.

Please try this:

	m_subscription.PublishingEnabled = true;

awcullen, 2018-01-25:
My suggestion is to use the 'defaults'.

```
// the DefaultSubscription has reasonable initial values.
Subscription m_subscription = new Subscription(session.DefaultSubscription);

// and so does DefaultItem.
var list = new List<MonitoredItem> {
                new MonitoredItem(m_subscription.DefaultItem){ ... }
};
```

vrushalikatkade, 2018-01-27:
thanks for reply , FastDataChangeCallback  is working
but after   m_subscription.Create(); function it throws the following exception
  "{"BadSessionIdInvalid"}"  same exception for OPCUAServer[index].Read() function
 
     m_subscription.FastDataChangeCallback = dtchange.FastDataChangeNotificationEventHandler;
                    _OPCUAServer[index].AddSubscription(m_subscription);
                    m_subscription.Create();

                    DataValueCollection values = null;
                    DiagnosticInfoCollection diagnosticInfos = null;

                    ReadValueIdCollection readvalue = new ReadValueIdCollection();

                    foreach(MonitoredItem mi in m_subscription.MonitoredItems)
                    {
                        ReadValueId rv = new ReadValueId();
                        rv.AttributeId = mi.AttributeId;
                        rv.NodeId = mi.StartNodeId;
                        rv.Handle = mi.Handle;
                        readvalue.Add(rv);
                    }          

                    ResponseHeader responseHeader = _OPCUAServer[index].Read(
                        null,
                        0,
                        TimestampsToReturn.Both,
                        readvalue,
                        out values,
                        out diagnosticInfos);

awcullen, 2018-01-27:
is _OPCUAServer[index].Connected ?

vrushalikatkade, 2018-01-27:
yes it is connected plz check the image

[image]

awcullen, 2018-01-27:
My test read values with no errors.

I added your lines:
```
            DataValueCollection values = null;
            DiagnosticInfoCollection diagnosticInfos = null;
            ReadValueIdCollection readvalue = new ReadValueIdCollection();

            foreach (MonitoredItem mi in subscription.MonitoredItems)
            {
                ReadValueId rv = new ReadValueId();
                rv.AttributeId = mi.AttributeId;
                rv.NodeId = mi.StartNodeId;
                rv.Handle = mi.Handle;
                readvalue.Add(rv);
            }

            ResponseHeader responseHeader = session.Read(
                null,
                0,
                TimestampsToReturn.Both,
                readvalue,
                out values,
                out diagnosticInfos);
```

to the NetCoreConsoleClient here:
https://github.com/OPCFoundation/UA-.NETStandard/blob/dbef06aa82abb94698d4c1e41458e8187180fdff/SampleApplications/Samples/NetCoreConsoleClient/Program.cs#L291

vrushalikatkade, 2018-01-29:
I have used this code in windows service not in console application.some tomes it will connected some time it throws exception

AlinMoldovean, 2018-01-29:
Hello @vrushalikatkade ,

That BadSessionIdInvalid error code should created and returned from the server side.
So you better check also the CreateSubscription request on server side.

AlinMoldovean, 2018-05-03:
closed due to inactivity..

closed 2018-05-03 by AlinMoldovean