OPCFoundation/UA-.NETStandard #336: How to use Quickstart Data Access Server and Client with HTTPS endpoints

RahulKate, 2018-01-29:
We are referring latest source code provided by OPCFoundation (Product Version : 1.03.351.0), 

We try to connect to "Quickstart Data Access Server" using "Quickstart Data Access Client" over https endpoints (https://desktop-7khbia4:62546/Quickstarts/DataAccessServer/), but we get below error while communicating.

Error-> BadUnknownResponse

[image]

[image]

[image]

Could you please check error or guide us with right direction?

Please send/redirect any manual if available for OPC UA samples to check over HTTPS.

AlinMoldovean, 2018-01-29:
Hello @RahulKate ,

The Https protocol support is not yet fully working and tested in this .NET Standard stack

RahulKate, 2018-01-29:
Thank you for response,

We have tried with earlier version 1.03.342.0, and its working successfully with https endpoints after following some steps given in below link to issue CA certificate (may be self signed certificate),
https://opcfoundation.github.io/UA-.NET/help/https_connectivity.htm

But its not working with recently released libraries from nuget or source code from github.

Is there any plan from OPC Foundation about the HTTPS support?

AlinMoldovean, 2018-01-29:
Hello @RahulKate ,

I think the problem is related to Issue #189

RahulKate, 2018-01-30:
Hi @AlinMoldovean ,

Thank you for your reply,

We saw #189, looks like its related to build problem of libuv.dll.

We will appreciate your views on below queries, based on your reply we have to publish about preferable solutions for OPC UA client.  

1) Is there any plan from OPC Foundation about the HTTPS support?
2) Which of below security parameters supported in OPC UA stack for **# opc:tcp** protocol?

- Security Modes: None, Sign, Sign & encrypt

- Security Policies: None, Basic128Rsa15, Basic256, Basic256Sha256

- Data Format: Binary, XML

3) Does OPC UA stack with Dot.Net Core (netstandard) build supports and tested for opc:tcp and https protocol with all security models?

RahulKate, 2018-02-05:
Hi @AlinMoldovean ,

Could you please reply/update on below queries?

We saw #189, looks like its related to build problem of libuv.dll.

We will appreciate your views on below queries, based on your reply we have to publish about preferable solutions for OPC UA client.

Is there any plan from OPC Foundation about the HTTPS support?
Which of below security parameters supported in OPC UA stack for # opc:tcp protocol?
Security Modes: None, Sign, Sign & encrypt

Security Policies: None, Basic128Rsa15, Basic256, Basic256Sha256

Data Format: Binary, XML

Does OPC UA stack with Dot.Net Core (netstandard) build supports and tested for opc:tcp and https protocol with all security models?

mregen, 2018-02-06:
@RahulKate I retested https which most recent bits and I found some issues. Fixes are here: #346.
Basically whats working:
- netcoreapp 1.1/2.0 app against 1.1/2.0
- .Net 4.6 https client may connect to any server. (Project may have to add System.Net.Http nuget package 4.3.3 to overcome runtime exception)
- .Net 4.6 https server still has the same issue. Its basically working, however Libuv.dll needs to be copied/added manually to the project and hence the endpoints must be setup in the config.xml (see #189)

From questions above:
- Which of below security parameters supported in OPC UA stack for # opc:tcp protocol?
- Security Modes: None, Sign, Sign & encrypt  --> All
- Security Policies: None, Basic128Rsa15, Basic256, Basic256Sha256  --> All
- Data Format: Binary, XML  --> depends, for transport only UA Binary, otherwise binary/xml/json
- Does OPC UA stack with Dot.Net Core (netstandard) build supports and tested for opc:tcp and https protocol with all security models?  --> yep

RahulKate, 2018-02-21:
Thank you for information,

We will check and OPC Foundation team about any help require in future

mregen, 2018-02-21:
see #359 how to enable https in a .Net app

closed 2018-02-21 by mregen