OPCFoundation/UA-.NETStandard #338: IOP Workshop 2018: Opc.Ua.SampleClient issues

mregen, 2018-02-02:
IOP Workshop 2018 follow up: Issues found by ITEI in Opc.Ua.SampleClient project (Universal Windows)

1.typo:
[image]

2. Click User, cannot enter Username/Password
[image]

3. Open new connection, click on user. White rectangle hides window forever.

[image]

[image]

AlinMoldovean, 2018-05-03:
we do not plan to fix issues on the controls for XAML or Winforms

xift, 2018-10-21:
I played around with this a little. 
The whole thing was obviously created as a proof of concept and never finished.
This project should be cleaned up (just remove the unfinished features from the UI) or deleted completely.

Regards,
Michael

closed 2019-02-21 by FischerSeb