OPCFoundation/UA-.NETStandard #339: IOP Workshop 2018: UA Sample Client cannot browse embedded server which limits MaxNodesRead to 20

mregen, 2018-02-02:
IOP Workshop 2018 follow up: Issues reported by ITEI with UA Sample Client (.Net 4.6) connecting to embedded UA server

2.8	The client should support to read part of nodes in the address space, because when the server has the limit per read, the client fails to read all the node.

UA Client connected, but doesn't show the server nodes

[image]

Wireshark exposes UA client reads 21 nodes per read

[image]

But the product reports operation limits as of: OperationLimits.MaxNodesPerRead.Value --> 20

Client should respect value and limit number of nodes read.

gillerman, 2018-02-02:
This is a topic to be discussed in Barcelona:

  *   Meta data (OPC UA Address spaces) need to be exchanged between systems
  *   Currently there are two ways for an OPC UA client to discover an OPC UA server address space
     *   Browse recursively through each server
        *   This has been shown to overload embedded servers because servers don’t know that this is more like an export that should be relegated to a background task
        *   Very chatty interface
     *   Export OPC UA address space from the server and Import into the client
        *   All or nothing approach does not allow the client to control what gets exported
  *   One possible solution:
     *   Create a standard program that takes a query filter as input parameter and returns a portion of the address space via the OPC UA wire protocol
     *   Allows servers to be aware that the client is treating the data more as an import

[image]<https://user-images.githubusercontent.com/7962757/35736056-08e22068-0827-11e8-8aa5-96894ffcc291.png>

[image]<https://user-images.githubusercontent.com/7962757/35736185-6191a6fc-0827-11e8-89d0-475d1af6e6b5.png>

closed 2019-02-21 by FischerSeb

FischerSeb, 2019-02-21:
no more reaction. Therefore we close this one.