OPCFoundation/UA-.NETStandard #347: How can i connect to DA server with dOPC or Matrikon ?

Talh4, 2018-02-07:
Hi,

    I'm trying to connect to Data Access Server (in the quickstar applications) with dOPC and Matrikon but i cant connect. I'm running the server and trying connect server with endpointurl and dOPC just waiting then timing out. When dOPC trying to connect and i shut down the server then dOPC saying server shutted down. So it know the server but can connect. Any suggestions ?

AlinMoldovean, 2018-02-07:
Hi @Talh4 ,

OPC UA and OPC DA (Classical OPC or COM OPC) are completely different communication protocol.
It is not possible for a Classical OPC Client to directly connect to OPC UA Sevrer.

Some gateway tools and aggregators are available as commercial products for that purpose.

Talh4, 2018-02-08:
First i was running an UA server and trying to connect from Matrikon and i got an answer that ua and da can not connect. Now i am running the DA server and trying to Connect from Matrikon and also dOPC and sitll can not connect. You are saying UA and DA can't connect okay but i am trying to connect DA to DA and still can't connect.

AlinMoldovean, 2018-02-08:
Hi, @Talh4 ,

If you are trying to run the "Data Access Server" sample from "SampleApplications\Workshop\DataAccess\Server", that is still an OPC UA Server and you need OPC UA Clients to connect to it.
It exposes the "Data Access" profile (or functionality) of OPC UA over the OPC UA protocol.
This repository does not contain samples for Classical OPC DA Server (or OPC COM ).
You can get OPC COM sample servers from commercial vendors or from here:
https://opcfoundation.org/developer-tools/developer-kits-classic/opc-classic-sample-servers/

Talh4, 2018-02-08:
@AlinMoldovean Oh now i see. In order to download the link you gave me. What kinda access do i need ? Because it says that i dont have permission for it.

AlinMoldovean, 2018-02-08:
@Talh4 The download is only for registered corporate members of OPC Foundation

Talh4, 2018-02-08:
@AlinMoldovean Thank you for your help.

closed 2018-02-08 by AlinMoldovean