OPCFoundation/UA-.NETStandard #348: TranslateBrowsePathsToNodeIds() with Simatic ProcessHistorian

fhurta, 2018-02-07:
I'm trying to use `TranslateBrowsePathsToNodeIds()` with ProcessHistorian (Simatic PCS 7 version 8.1). The method works well for `Server` node but for node `Process Historian {GUID}` it always fails with `BadNoMatch` status code.

I described the details [here](https://stackoverflow.com/questions/48565658/translatebrowsepathstonodeids-opc-ua-method-with-simatic-processhistorian).

Could it be issue in OPC UA library or rather limitation of ProcessHistorian? Any experience with it?

What I need is to be able to translate browse paths for tags to their NodeId.

AlinMoldovean, 2018-02-07:
Hello @fhurta ,

If the relative path was created correctly, the problem seems to be on the server side.
"BadNoMatch" is normally returned when the when the node cannot be find by the server.
So you better check with the server vendor.

fhurta, 2018-02-09:
Just found out the ProcessHistorian officially supports only `Browse` and `BrowseNext` services of `View  Service Set`.

So this issue can be closed, thanks.

closed 2018-02-09 by AlinMoldovean