OPCFoundation/UA-.NETStandard #349: Can UA-.NETStandard be built without NuGet packages?

SoftwareDave, 2018-02-07:
The UA-.NETStandard requires several NuGet packages to be installed, and that's installed per application, not just a one time install as DLLs or libraries. That's over 650MB of extra overhead per application.

**Question:** Is there a way to install and run the UA-.NETStandard package and compile/build the DLLs and EXEs without the cumbersome NuGet packages and dependencies? That's why we chose to go with the UA-.NET-master.zip (here after now called the UA-.NET-Legacy-master.zip) - it has no NuGet dependencies.

Thanks - Dave.

mregen, 2018-02-13:
@SoftwareDave thanks for the heads up, I found a few improvements wrt to the nuget packages --> Please try this: https://github.com/mregen/UA-.NetStandardLibrary/commits/clean_nuget
However all the other dependencies are mandatory, unless you get rid of https server (kestrel) & Newtonsoft JSON. Whats your work flow to get to 650 MB? I see bin folders in the 10's of MB but not in the hundreds?

Cheers,
Martin

SoftwareDave, 2018-02-13:
Thanks, mregen:

First, I had to install 3 NuGet packages:
1) the  NETStandard.Library v2.0.1
2) the OPCFoundation.NetStandard.Opc.Ua v0.4.5
2) the OPCFoundation.NetStandard.Opc.Ua.Sumbols v0.4.5

Then, as a test, I created a C# Windows Forms application:  I simply tested the Utils.Trace method with a single button.  The application has a reference to Opc.Ua:   using Opc.Ua;

using Opc.Ua;

namespace WindowsFormsApp2017
{
    public partial class Form1 : Form
    {
        // private properties
        private string  m_outputFilePath;
        private string  m_UtilsLogFilePath;
        private bool    m_deleteOnLoad  = true;
        private int     m_traceMasks    = Utils.TraceMasks.Error | Utils.TraceMasks.Information;  

        public Form1()
        {
            InitializeComponent();
        }
        private void Form1_Load(object sender, EventArgs e)
        {
            // setup the Utils Trace log file.
            m_outputFilePath = System.IO.Path.GetDirectoryName(Application.ExecutablePath);
            m_UtilsLogFilePath = m_outputFilePath + "\\WindowsFormsApp2017.log.txt";
            Utils.SetTraceLog(m_UtilsLogFilePath, m_deleteOnLoad);
            Utils.SetTraceMask(m_traceMasks);
        }
        private void TestButton1_Click(object sender, EventArgs e)
        {
            // do some stuff.
            Utils.Trace(Utils.TraceMasks.Information, "TestButton1_Click: Hello OPC UA!");
        }
       
    }
}

The .\packages\  folder is about 390MB for this simple application.

One of our other Client test applications had 490MB in the packages folder, and one of the servers had about 650MB.

That amount of overhead, per application, makes it unusable for us.  Or maybe my installation and application creation methodology is flawed?  Thanks for any insights you have on this.  
-Dave

mregen, 2018-02-14:
Hi @SoftwareDave, normally nuget is configured to use a shared packages folder under Userhome/.nuget/packages. see here: https://docs.microsoft.com/en-us/nuget/reference/nuget-config-file. If this setting is missing you can add it with `nuget config`. And yes, indeed the packages folder tends get quite large. Hope this helps..

mregen, 2018-02-14:
Anyway, it seems like a good idea to clean up nuget dependencies: PR --> #353

SoftwareDave, 2018-02-14:
Indeed  - in order to meet our software standards (version control, build
control, security) - we would need a NuGet-free library;  thus we prefer
the UA-.NET-master.zip (here after now called the
UA-.NET-Legacy-master.zip).  We can put all our DLLs in one folder for all
of our Client, Server, and Library projects and solutions to reference.
And it makes the Setup/Install easier.

Thanks for creating this new issue #353 - I have subscribed to it.
-Dave

mregen, 2018-02-15:
Hi @SoftwareDave, to set expectations right there will *not* be a version without nuget packages... #353 is just trimming down a bit.
However you can configure your nuget feed to take only local packages. This way you can use source control and make sure only approved packages are used..

SoftwareDave, 2018-02-15:
Thanks - I'll try out the new release when it's available.  -Dave

mregen, 2018-02-17:
#353 merged

closed 2018-02-17 by mregen