OPCFoundation/UA-.NETStandard #356: BadUserSignatureInvalid when authenticating with X509 Certificate

MaartenBE, 2018-02-14:
I am writing a client that connects to a server implemented with the Unified Automation C#.NET. For user authorization the server requires X509 UserIdentity token. 

When I try to use a valid X509 certificate to authorize the user I receive a "BadUserSignatureInvalid" in the ActivateSessionResponse.

After researching the error, the issue seems to be that the ServerCertificate sent back from the server contains a chain. The OPC UA Specification version 1.04 part 4 - 5.6.3.2 states: 

> If the serverCertificate contains a chain, the signature calculation shall be done only with the leaf Certificate. For backward compatibility a Server shall check the signature with the full chain if the check with the leaf Certificate fails.

The current client implementation takes the full certificate including the chain. To be compliant, the client should only send the leaf certificate in the ActivateSessionRequest. 

If we can agree that this is required, I can maybe create a pull request with the implementation.

mregen, 2018-02-14:
Hi @MaartenBE, cert chain support is required and any contribution is welcome!

MaartenBE, 2018-02-15:
I have created pull request #358. With this commit a X509Certificate2 is created from the certificate data Byte[] received from the ActivateSessionResponse. The RawData property, which only includes the leaf of the certificate chain is then used for signature generation.

AlinMoldovean, 2018-02-19:
Hello @MaartenBE ,

Could you provide access to the server used for testing?
According to the specification, the server should also accept signatures with the complete chain, for backward compatibility:
   _"For backward compatibility a Server shall check the signature with the full chain if the check with the leaf Certificate fails."_

We will soon implement support for certificate chains in #362 and will include also a fix for this.
Anyway, the problem seems to be on validation the "userTokenSignature" field of the ActivateSesion() request (not "clientSignature").

Thanks,
Alin

MaartenBE, 2018-03-01:
Yes, the server implementation is not compliant to OPC UA spec 1.04 where this clarification was added. I have reported this to Unified Automation.

I cannot provide you with the real server implementation, since it is a custom development of another company. It is using the C#.NET SDK from Unified Automation so you can download the evaluation version of the SDK from the Unified Automation website after registration.

For my project I have a workaround and the fix will then be a part of #362 so I will close this issue.

closed 2018-03-01 by MaartenBE